TcMenu/tcMenu #283: Designer: once saved, focus goes to root

luwenjin, 2023-01-13:
If I was editing an menu, and saved the project, the designer will automatically select the root, and I had to select back the menu I was editing.

I think, saving shouldn't change the selected menu.

davetcc, 2023-02-17:
As per the other chat, this does not happen on macOS Ventura or Windows 11, can you please provide the environment as there's something different here.

The focus returns to the tree (not root but the last selected item), but simply pressing Esc goes back to the editor.

luwenjin, 2023-02-20:
I've tried it on ventura+designer 3.1

the simpleU8g2.emf from the examples works fine. 
But my emf has the bugs I've posted, I had also looked up the logs, found no errors, here is a sample from the log file: 
```
2023-02-20 09:28:22 信息 com.thecoderscorner.menu.editorui.project.FileBasedProjectPersistor save Save file starting for: /Users/luwenjin/Projects/luwenjin/SoccerBar/src/ui/SoccerBar.emf
2023-02-20 09:28:22 信息 com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl createPanelForMenuItem creating new panel for menu item editing 0
2023-02-20 09:28:23 信息 com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl createPanelForMenuItem creating new panel for menu item editing 85
2023-02-20 09:28:23 信息 com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl lambda$createPanelForMenuItem$3 created panel UITextMenuItem
2023-02-20 09:28:24 信息 com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl createPanelForMenuItem creating new panel for menu item editing 93
2023-02-20 09:28:24 信息 com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl lambda$createPanelForMenuItem$3 created panel UIBooleanMenuItem
2023-02-20 09:28:26 信息 com.thecoderscorner.menu.editorui.project.FileBasedProjectPersistor save Save file starting for: /Users/luwenjin/Projects/luwenjin/SoccerBar/src/ui/SoccerBar.emf
2023-02-20 09:28:26 信息 com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl createPanelForMenuItem creating new panel for menu item editing 0
2023-02-20 09:28:27 信息 com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl createPanelForMenuItem creating new panel for menu item editing 93
2023-02-20 09:28:27 信息 com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl lambda$createPanelForMenuItem$3 created panel UIBooleanMenuItem

```

I tried to trim my project down to see which branch cause the bugs, 
Here is the minimal project that can reproduce the bugs.

[bug.emf.zip](https://github.com/davetcc/tcMenu/files/10778677/bug.emf.zip)

davetcc, 2023-03-27:
As per the other issue this is not recreatable on either my Windows or MacOs based machine. Can you provide more detail, I'm measuring this by the selected item in the tree's UI, are you using a different measure?

luwenjin, 2023-03-30:
Sorry, I don't understand the meaning of "measure" in the context, could you please express it in another way?

Here is a screen recording of the bug.

https://user-images.githubusercontent.com/251550/228581173-11ce30ca-5abd-4eda-9182-b773f72f93fb.mp4

davetcc, 2026-09-06:
Web designer is now the standard designer, I'll close this out for now.

closed 2026-09-06 by davetcc