TcMenu/tcMenu #292: Menu/EEPROM: Factory reset support

vzahradnik, 2023-02-04:
Currently it is not possible to reset the menu to its factory state:
- There is no method to reset the menu, including menu item values
- Item is assigned a default value after bootup, but once the item value is changed, we can't read the default value anymore

There is a workaround, which involves writing a custom code to set each menu item to a default value and then saving the config. However, in this code we duplicate the information stored inside our project file. For example:
  - We set the default value in TcMenu Generator
  - We set the same default value in our custom function to factory reset the menu

davetcc, 2023-02-05:
To factory reset, I normally just erase the magic key in the first two bytes and reset the device. When it comes back up the EEPROM will not load, and everything will be at the default values as if it were starting for the first time. In fact, I generally wouldn't even do the reset, just tell the user to reset in case it was not convenient.

For my case of mainly audio equipment and a few garden devices, this works well for this very occasional case.

There is also a change going into 3.1 that will allow for sized EEPROM, IE the largest location saved out is also stored, and the loader will ignore any item with a location above that. This allows for default values in new items.

vzahradnik, 2023-02-06:
Yeah, I saw you already updated the docs but the code is not yet available. I reserved two bytes already to enable this feature.

Ok, your suggestion makes sense. Device reset is rather quick.

Thanks!