TcMenu/tcMenu #293: SubMenuItem: Callback support

vzahradnik, 2023-02-04:
I have a submenu where I dynamically pull the data from the sensors; it's called service screen.

I'd like to define a callback function which is called after user enters the menu. Currently, `SubMenuItem`s don't support callback functions.

My workaround is to periodically check whether user is or is not in this menu:

```cpp
    if (menuServiceMenu.isActive()) {
        ...
    }
```

This code works however is not efficient. Instead of executing the code only once (as in the case of callback version), it is called periodically while user stays in the screen.

davetcc, 2023-02-04:
https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/menumanager-and-iteration/#listening-for-changes-on-menu-manager

Can you take a look and see if the menu manager observer does what you want. It sounds like it should work for you and you can even prevent the menu from showing if you wanted.

davetcc, 2023-02-06:
In fact, if you could take a look at the below as it does pretty much what you want: 

https://github.com/davetcc/tcMenuLib/blob/master/examples/arduino32/dynamicMenuItems/dynamicMenuItems.ino#L48

vzahradnik, 2023-02-13:
Your solution works great. I'll close this issue. I just have one question. What is the purpose of `menuMgr.getCurrentMenu() == item`? When I had this clause in the if statement, my code was not triggered. Once I removed this part, my code started working.

```
if(item->getId() == menuList.getId() && menuMgr.getCurrentMenu() == item) {

}
```

davetcc, 2023-02-13:
The examples sometimes go overboard with things to show all the possible functions that you can use. Checking by ID is fine.

closed 2023-02-13 by vzahradnik