TcMenu/tcMenu #297: designer bug: in BooleanMenuItem, options under"Default Value" doesn't reflect the choice of "Possible values"

luwenjin, 2023-02-07:
[image]

luwenjin, 2023-02-14:
The Default Value's options is not updated instantly, but after switching to other menu and back, the options is updated according to Possible responses

davetcc, 2023-02-17:
I've tested this on Windows 11 and macOS Ventura and can't recreate this behaviour in 3.1.

Do you use a different environment to that? Is there anything specific about the project? IE could you try this with an example using the 3.1 UI. If you can please look in the log (~/.tcmenu directory) and see if there are any errors.

luwenjin, 2023-02-20:
I've tried it on ventura+designer 3.1

the simpleU8g2.emf from the examples works fine. 
But my emf has the bugs I've posted, I had also looked up the logs, found no errors, here is a sample from the log file: 
```
2023-02-20 09:28:22 信息 com.thecoderscorner.menu.editorui.project.FileBasedProjectPersistor save Save file starting for: /Users/luwenjin/Projects/luwenjin/SoccerBar/src/ui/SoccerBar.emf
2023-02-20 09:28:22 信息 com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl createPanelForMenuItem creating new panel for menu item editing 0
2023-02-20 09:28:23 信息 com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl createPanelForMenuItem creating new panel for menu item editing 85
2023-02-20 09:28:23 信息 com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl lambda$createPanelForMenuItem$3 created panel UITextMenuItem
2023-02-20 09:28:24 信息 com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl createPanelForMenuItem creating new panel for menu item editing 93
2023-02-20 09:28:24 信息 com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl lambda$createPanelForMenuItem$3 created panel UIBooleanMenuItem
2023-02-20 09:28:26 信息 com.thecoderscorner.menu.editorui.project.FileBasedProjectPersistor save Save file starting for: /Users/luwenjin/Projects/luwenjin/SoccerBar/src/ui/SoccerBar.emf
2023-02-20 09:28:26 信息 com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl createPanelForMenuItem creating new panel for menu item editing 0
2023-02-20 09:28:27 信息 com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl createPanelForMenuItem creating new panel for menu item editing 93
2023-02-20 09:28:27 信息 com.thecoderscorner.menu.editorui.uimodel.CurrentProjectEditorUIImpl lambda$createPanelForMenuItem$3 created panel UIBooleanMenuItem

```

I tried to trim my project down to see which branch cause the bugs, 
Here is the minimal project that can reproduce the bugs.

[bug.emf.zip](https://github.com/davetcc/tcMenu/files/10778677/bug.emf.zip)

davetcc, 2023-02-20:
Thanks it must be environment specific. I’ll load the project and try myself.

davetcc, 2023-03-18:
I unfortunately cannot recreate this locally, even with the same project.

luwenjin, 2023-03-21:
Not a big issue, let's see if anyone else encounters the problem, please close the issue for now.

closed 2023-03-21 by davetcc