TcMenu/tcMenu #302: Designer: Allow to move SubMenuItems up or down in the menu hierarchy

vzahradnik, 2023-02-16:
I'd like to move whole SubMenuItem groups into other SubMenus either down the hierarchy or to the parent level. Currently the only way known to me (besides recreating the menu from scratch) is to manually edit the project file.

Use case:

*From:*
```
Settings -> Display -> ...
         -> Date and Time -> ...
         -> Network -> ...
...
 ```

*To:*
```
Display -> ...
Date and Time -> ...
Network -> ...
...
 ```

davetcc, 2023-02-17:
Yep, agreed, copy and paste works a bit oddly now for whole sub menu blocks in the tree, copy the submenu (either Ctrl-C or from menu), and paste it back in (either Ctrl-V or from menu). The problem is that it pastes it back in the same place. I'll solve this when I look at drag and drop (which is easier than I expected).

davetcc, 2023-03-26:
The direct move buttons have been removed, and it is now possible to drag and drop items easily anywhere in the structure, this even works with undo and redo. See #303

closed 2023-03-26 by davetcc