TcMenu/tcMenu #306: Menudesigner 3 OS X

tomelgato, 2023-02-19:
On some input fields (in my case Menu Variable Name) with long text the cursor disappears at the end under the x, you dont see the last 2-3 characters while editing
[image]

luwenjin, 2023-02-20:
Me too, might be blocked by the X icon

davetcc, 2023-02-20:
Yep, it seems to be an issue with the theme we are using, I'll need to see if I can adjust the theme to remove this annoyance (or make the boxes bigger so it rarely happens. One of the two.

tomelgato, 2023-02-28:
Bigger boxes would be no good solution. How big should they be? I have rather long names due to nested menus so it would not work in this case or you need to do annoying long boxes which are not needed in most of the cases.

davetcc, 2023-03-26:
These are coming from the theme we are using, I suspect I'm doing something that is throwing off the theme, but it is going to need me to create a standalone app with the theme and see what happens there, and then if need be raise a bug in the source theme library. Moving out of 3.2 for now.

tomelgato, 2023-05-29:
Bug still exists in 4.0

davetcc, 2023-05-29:
I thought the answer to this was going to be really complex, but it turns out there is a really easy way, in the global style I just tuned off the right button, it seems to work. Based on https://github.com/JFXtras/jfxtras-styles/issues/180

It will be in 4.1

davetcc, 2023-05-29:
[image]

See above no cross icon on right.

closed 2023-05-29 by davetcc