TcMenu/tcMenu #310: TCmenu 3 sub menus

tomelgato, 2023-02-28:
I have a menustructure with submenus. When i go into a submenu i dont see a cursor. I need to go down, then the cursor appears in the 2nd line (1st item of submenu). I need to go up again to go to the submenutitle to be able to go up in menustructure again (before i dont see the [..] too).
Similar behavior when reaching the end of the submenu and jumping to 1st item again

tomelgato, 2023-02-28:
Sorry, youre right. Will do a video next time when im at the machine.

4x20 LCD with 4x5 matrix keyboard. It happens at the first sub level already. Its like the cursor is deactivated when you enter the submenu and it gets shown after the first valid keystroke

davetcc, 2023-03-01:
Even a screen shot of the situation with a description would be helpful. 
Thanks