TcMenu/tcMenu #312: Adafruit_SPITFT error with ST77xx screen

THAWINKELS, 2023-02-28:
When generating an code for an TFT ST7735 screen my Arduino IDE gives the error @file tcMenuAdaFruitGfx.h

tcMenuAdaFruitGfx.h:108:26: error: variable or field 'drawCookieCutBitmap' declared void
 void drawCookieCutBitmap(Adafruit_SPITFT* gfx, int16_t x, int16_t y, const uint8_t *bitmap, int16_t w,
                          ^~~~~~~~~~~~~~~

Does anyone know where this error is from?

davetcc, 2023-02-28:
Have you checked all libraries and the UI are up to date? Please check all versions. What is the environment that you are using?

THAWINKELS, 2023-03-01:
Hi,

All the libary's are updated to the last (see picture)
[image]

I'm working om Windows 8 x64, Arduino IDE with Arduino UNO.

Adafruit_GFX libary is on version 1.11.5

[image]

davetcc, 2023-03-01:
Graphical displays other than OLED using the SSD1306Ascii are not supported on Uno, there is not enough memory.

Even if you got past the compile error, it would not link due to a lack of available memory. The smallest 8-bit board that works for graphical displays is MEGA2560.

davetcc, 2023-03-01:
OK, can you please zip up a test project and attach it so I can look at what's got generated?

davetcc, 2023-03-01:
BTW - Uno works well with LCD displays that are wired directly (EG DfRobot shield for example) and also for OLED using SSD1306Ascii. We even have products running live on Uno with a direct LCD connection!

davetcc, 2023-03-01:
Ah yes, the old IDE does not update, you often need to close and reopen the project if a file changes outside. The new 2.0 IDE and platformIO do not have this limitation.

THAWINKELS, 2023-03-02:
Hey,

I'm currently testing with an arduino mega2560 but the error is back, I only changed the pins in the code generator.
See attachment.

[DisplayMenuTest.zip](https://github.com/davetcc/tcMenu/files/10875048/DisplayMenuTest.zip)

THAWINKELS, 2023-03-02:
[image]
Adafruit_ST7735 gfx(10, 9, 8);   (libary is up to date)
Gives the error

When I change the display type:
[image]

Adafruit_ILI9341 gfx(10, 9, 8);
There is no error.

So something with the ST7735 libary is wrong.
**I deleted all ST7735 libarys and install them again, error is gone.**

THAWINKELS, 2023-03-02:
[image]

When I load the sample program (change some pin settings) I don't see the right menu text,
Do I miss something?

Inside the menu, the text is better:
[image]

davetcc, 2023-03-03:
What platform is selected in designer, that looks like a progmem issue. It must be set to AVR for Mega and Uno.

davetcc, 2023-03-06:
Can we close this now?

closed 2023-03-08 by davetcc

davetcc, 2023-03-13:
Assuming this was fixed by moving to using the right platform? If not please reopen.