TcMenu/tcMenu #321: Bug: AdafruitDrawable's internalTextExtents does not return magnitude-scaled height value when using Font

NachtRaveVL, 2023-03-23:
At the end of the `Coord AdafruitDrawable::internalTextExtents(const void *f, int mag, const char *text, int *baseline)` method that is generated in the code generator, the height values return in the case of a font being present is not correctly scaled by the passed magnitude value. This also is the case for the returned baseline value.

A simple enough fix:
```
        if(baseline) *baseline = (bl * mag);
        return Coord((int)w, (height * mag));
```

Although I would also suggest to change `const char sz[]` to PROGMEM, thus not taking up RAM space for a random string that is only used to generate the height value. I would also suggest using a cache value here, as the chances of subsequent calls with the exact same parameters (once to do the sizing/positioning, again to do the rendering) is highly likely and a single pointer set up to operate as an LRU cache would speed up rendering code.

I've made all these improvements (minus LRU cache) in the following updated method's else statement:
```
    else {
        // we need to work out the biggest glyph and maximum extent beyond the baseline, we use 'Ay(' for this
        static const char sz[] PROGMEM = "AgyjK(";
        int height = 0;
        int bl = 0;
        const char* current = sz;
        auto fontLast = pgm_read_word(&font->last);
        auto fontFirst = pgm_read_word(&font->first);
        while(pgm_read_byte(current) && (pgm_read_byte(current) < fontLast)) {
            size_t glIdx = pgm_read_byte(current) - fontFirst;
            auto allGlyphs = (GFXglyph*)pgm_read_ptr(&font->glyph);
            unsigned char glyphHeight = pgm_read_byte(&allGlyphs[glIdx].height);
            if (glyphHeight > height) height = glyphHeight;
            bl = glyphHeight + (signed char)pgm_read_byte(&allGlyphs[glIdx].yOffset);
            current++;
        }
        if(baseline) *baseline = (bl * mag);
        return Coord((int)w, (height * mag));
    }
```

Personally I think the whole baseline thing should be a pre-computed value for any given font. Doing this sort of operation at runtime is highly unnecessary and a single uint16_t would be enough to store all the needed extra data that would avoid doing all of this (often times *during* the render loop). Definitely a time/space trade-off situation.

davetcc, 2023-03-23:
When using tcUnicode it does cache the baseline and height ahead of time, but it cannot calculate text sizes ahead of time, and that is the bigger cost. I could probably make native adafruit do the same thing (or I could move native adafruit to draw using tcUnicode). Either is possible but it all takes time.

NachtRaveVL, 2023-03-23:
Yup, the text sizes do have to be recalculated, but only if the parameters change. Guaranteed half your calls into this costly method are followed up by the same call with the same parameters (well, I know because it's in the debugger). Luckily you can use a simple associative array (or some other mechanism) to pretty much resolve a lot of that business, but yeah. Probably shouldn't be scanning for largest font size when you can cache that ahead of time and skip all that. Pair that with a simple 1-time LRU cache and it'll speed all the rendering code up, for sure.

davetcc, 2023-03-23:
As per the last comment - When using tcUnicode it does cache the baseline and height ahead of time. We could look to do the same there for native font rendering.

Prioritizing performance over memory, many using the Adafruit library want to do the opposite as they are on small boards. 

Further, even implementing a cache without STL would be quite complex (not all boards support STL or have space to enable it). Further, a cache lookup, even if map based, will most likely not be faster than array traversal. The hashing/searching of the input string alone in the cache will cost more than the current code.

davetcc, 2023-03-23:
And don't forget the largest part of the operation is updating the screen. With Adafruit this is done mainly by the processor copying around memory and then synchronously calling the SPI functions. Using the buffering (sprite) support in TcMenu makes it as fast as it can be, but it is still orders of magnitude slower than anything else. 

That is why many choose to use TFT_eSPI on 32-bit hardware, it has DMA/Async support on bigger boards.

NachtRaveVL, 2023-03-23:
Yup, I agree that there is a complexity thing at work there - but you have SimpleCollections pulled in already, no? Could just use it for something basic like an associative container, don't need to do full STL. In fact what I'm doing for backwards STL support is using ArxContainer, and it doesn't even do a proper associative map - everything is a bi-directional array with static size. Maps are made into Pair<> objects. I mean, it works - AVRs have no memory for that sort of thing, so. But I think just having a simple 1-item LRU is going to speed things up immensely. I did something very similar to the string return method we have in ours and it was a massively huge speed-up in terms of reduced external access, especially in the case of externalized library data.

davetcc, 2023-03-23:
The string comparisons (to ensure the size returned was the right one) would take nearly as long as the font size calculation, especially if the baseline part is removed. There is no advantage, it would probably be slower, and far more complex.

davetcc, 2023-03-23:
At the end of the day, if you think there is significant improvement why not try timing it after I've made the change to cache the baseline, if there's a large difference, pingback here, and we'll look at it again, see if some kind of option should be added. Maybe added to tcUnicode and made generic enough for everything to use.

davetcc, 2023-04-16:
I've done the caching of the last baseline, if the font does not change, it does not get recalculated. It will hopefully get enough testing on the various board I'll use during 3.2 testing to be sure all is good.

davetcc, 2023-04-16:
Tested and working on the pico Adafruit example

closed 2023-04-16 by davetcc