TcMenu/tcMenu #323: [I18N] Designer: Allow users to add custom locale strings and expose them in C++ code

vzahradnik, 2023-03-23:
I just bumped into a usecase which seems to me generic enough to have some kind of support in the library.

Basically, TcMenu will support menu translations in the upcoming release. But the support is implemented (as far as I know) only for menu items defined in the project file.

Now imagine a scenario that we want to display a dialog with some text in it. At the moment, the string needs to be defined in place of the function call:

```
 dlg->copyIntoBuffer("Some text to be translatable.");
```

Sure, we could create defines similarly to what TcMenu does but this will break app strings to two places - one in the properties file and the other in the C++ code.

Another scenario, albeit not that visible, is logging or printing some kind of output to the Serial. By moving the strings into a common place, these lines could also be localized, if necessary.

Here is my proposal:
- Allow users to directly edit the property file with their arbitrary text editor (i.e., support in the Designer UI is optional but nice to have, like a simple table with the options to Add/Delete strings)
- Generate C++ code and include these custom strings as well. As long as users stick to the format defined by TcMenu, I don't see any problem with exposing these extra fields.

Thanks for considering this.

davetcc, 2023-03-24:
That is pretty much how it works

davetcc, 2023-03-25:
> Basically, TcMenu will support menu translations in the upcoming release. But the support is implemented (as far as I know) only for menu items defined in the project file

Actually, the support adds resource bundle properties into the menu structure. The designer can out of the box handle translations for names, enums and units. But, the properties files are text, they are written to in a way that is non-destructive, so you can add as many additional items as you want. 

The output is a series of C++ header files that have entries based on what was in the properties, it will have all entries that are present in the default bundle, specialized by country and locale if available.

For formatting, that would be user level, if someone wants to raise a PR for an "extra" that would be great, but it must not be in the core, as this has to work on small boards.

davetcc, 2023-03-25:
You'll get the chance to try it soon, as soon as I get the drag and drop working I'll be making a build. Maybe in another week or two. I've really slowed down on this now, and going forward, I want to draw a line under where designer is for now and move to a bug fixing phase, I personally think it is now pretty feature complete but needs a bit of tidying up.

 I'll help with the mobile app and external APIs as I need both of these in a project that I've put on hold to do the i18n work.

davetcc, 2023-03-25:
[image]
The inbuilt support just allows the locale to be changed using the drop-down, if an item doesn't need language level overrides, you can set an item not to have locale entries.

davetcc, 2023-03-25:
You can take a look at how it works on master of the library, you'll see the properties files in the i18n directory and the output language files in the generated directory:

https://github.com/davetcc/tcMenuLib/tree/master/examples/arduino32/picoAdafruitDashboard

davetcc, 2023-03-26:
Going to close as already catered for in the design, if you disagree after trying the BETA please do reopen.

closed 2023-03-26 by davetcc