TcMenu/tcMenu #324: Dialog with menu items render issues

vzahradnik, 2023-03-23:
**Describe the bug**
I implemented a custom multiline dialog following this [thread](https://www.thecoderscorner.com/jforum/posts/list/134.page). I have observed two issues:
- When I go up and back to the cancel button, next to the button is rendered part of the text from the item above it.
https://www.youtube.com/watch?v=9fgybk6Pqb0
- Not the whole screen is used. As it's shown on the picture, the whole dialog can fit the screen but it is shown like this only after I scroll to the top of the dialog.
[image]

**Commerical or Personal use**
Commercial use.

**To Reproduce**
Steps to reproduce the behavior, if a UI issue please detail all steps and include an example project if needed.
If a library issue provide whatever is needed in order for me to quickly identify the cause.
```cpp
RENDERING_CALLBACK_NAME_INVOKE(fnVendorLine, textItemRenderFn, "", -1, NO_CALLBACK)
TextMenuItem vendor(fnVendorLine, nextRandomId(), 20, nullptr);
TextMenuItem website(fnVendorLine, nextRandomId(), 20, &vendor);
char aboutStr[20];
char versionStr[20];
char vendorStr[20];

void CALLBACK_FUNCTION ShowProductInfo(int id) {
    withMenuDialogIfAvailable([](MenuBasedDialog* dlg) {
        dlg->setButtons(BTNTYPE_NONE, BTNTYPE_CLOSE);

        vendor.setReadOnly(true);
        website.setReadOnly(true);

        sprintf(aboutStr, "About %s", TOSTRING(PRODUCT_NAME));
        sprintf(versionStr, "Version: %s", TOSTRING(PRODUCT_VERSION));
        sprintf(vendorStr, "Vendor: %s", TOSTRING(VENDOR));

        vendor.setTextValue(vendorStr);
        website.setTextValue(TOSTRING(VENDOR_WEBSITE));

        dlg->insertMenuItem(&vendor);
        dlg->insertMenuItem(&website);

        GridPosition vendorGridPlacement(GridPosition::DRAW_TEXTUAL_ITEM, GridPosition::JUSTIFY_LEFT_VALUE_ONLY, 2);
        renderer.getDisplayPropertiesFactory().addGridPosition(&vendor, vendorGridPlacement);

        GridPosition vendorWebsiteGridPlacement(GridPosition::DRAW_TEXTUAL_ITEM, GridPosition::JUSTIFY_LEFT_VALUE_ONLY, 3);
        renderer.getDisplayPropertiesFactory().addGridPosition(&website, vendorWebsiteGridPlacement);

        dlg->show(aboutStr, false);
        dlg->copyIntoBuffer(versionStr);
    });
}
```

**Embedded environment
 Custom HW board with ESP32 and U8G2 monochrome display (`U8G2_SSD1309_128X64_NONAME0_F_HW_I2C`).

davetcc, 2023-03-24:
Thanks for the detailed report. As soon as I get the card layout and Unicode beta sorted I’ll have a look.

I think it is because the item is maybe a pixel too big and the way rendering works is to ensure the item is not obscured from view. I’ll have to try this case myself.

vzahradnik, 2023-03-24:
I can help with getting more information if needed. Just let me know.

davetcc, 2023-04-16:
This is possibly a little complex, I think what's happening is that the renderer is calculating the row order before the item is added somehow, or not properly taking into account the added item. I think I've seen this before but could not recreate it, this seems easier to reproduce, so I'll take a good look at this after 3.2 and debug exactly what's causing it.

vzahradnik, 2023-04-16:
OK, if there is a way how I could help, e.g. by testing your changes, let me know.

davetcc, 2023-07-29:
Part 1 of the detangling of the rendering layer is now completed. I've basically taken out all attempts at lazy loading, as I'm not sure what good they actually did anyway.

davetcc, 2023-08-10:
Part 2 is now done as well, I'm going to open a discussion around this to explain the changes and rationale.

davetcc, 2023-08-10:
https://github.com/davetcc/tcMenu/discussions/387

davetcc, 2023-08-17:
I'm going to leave this open, but I believe that if you were to use the latest master version of the library there is a very good chance many of these rendering issues may already be fixed.

davetcc, 2023-08-28:
@vzahradnik - can you check if all the issues are fixed in 4.1.1 library? If there's nothing outstanding please close.

vzahradnik, 2023-08-28:
Hi Dave, I will. But I don't have the development board from this project at hand.

Let's leave it open and as soon as I will be able to test it out, I will update/close this issue.

Thanks!

davetcc, 2023-09-05:
So in short, most rendering issues are now fixed, but the card layout issue you saw with a later item wrongly showing first is probably because you started your grid positions at row 1 instead of row 0. You should only start at row 1 when there is a title, otherwise, row 0 will be unoccupied and taken by the first item that does not have a grid row.

I'm actively updating documentation at the moment and will make that clearer.

Going to close for now, please reopen if you recreate any of the issues

closed 2023-09-05 by davetcc