TcMenu/tcMenu #342: [i18n] Analog item with no unit - TcMenu throws exception

vzahradnik, 2023-04-30:
**Describe the bug**
This issue occurrs in the experimental build with i18n support. After the feature is activated, TcMenu designer doesn´t allow to click on the analog menu item. Instead it logs an exception. There is no crash but the designer window is not switched to the selected menu item.

**Commerical or Personal use**
Commercial use.

**To Reproduce**
Steps to reproduce the behavior, if a UI issue please detail all steps and include an example project if needed.
If a library issue provide whatever is needed in order for me to quickly identify the cause.

Here is the problematic config:
```
    {
      "parentId": 71,
      "type": "analogItem",
      "defaultValue": "0",
      "item": {
        "maxValue": 65535,
        "offset": 0,
        "divisor": 1,
        "unitName": "",
        "step": 1,
        "name": "Photoresistor",
        "variableName": "PhotoresistorValue",
        "id": 72,
        "eepromAddress": -1,
        "readOnly": true,
        "localOnly": false,
        "visible": true,
        "staticDataInRAM": false
      }
    },
```

As I found out, the problem occurrs only when the `AnalogMenuItem` has no unit assigned. In this case I read raw sensor value without any processing.

Here is the problematic code:
```java
default String getFromLocaleWithDefault(String localeEntry, String defText) {
        if(!isLocalSupportEnabled() || !localeEntry.startsWith("%") && localeEntry.length() > 1) return defText;
        if(getCurrentLocale().getLanguage().equals("--")) return defText;
        String ret = getLocalSpecificEntry(localeEntry.substring(1));
        return (ret == null) ? defText : ret;
    }
```

And the full error stack trace:

```java
Exception in thread "JavaFX Application Thread" java.lang.StringIndexOutOfBoundsException: Range [1, 0) out of bounds for length 0
	at java.base/jdk.internal.util.Preconditions$1.apply(Preconditions.java:55)
	at java.base/jdk.internal.util.Preconditions$1.apply(Preconditions.java:52)
	at java.base/jdk.internal.util.Preconditions$4.apply(Preconditions.java:213)
	at java.base/jdk.internal.util.Preconditions$4.apply(Preconditions.java:210)
	at java.base/jdk.internal.util.Preconditions.outOfBounds(Preconditions.java:98)
	at java.base/jdk.internal.util.Preconditions.outOfBoundsCheckFromToIndex(Preconditions.java:112)
	at java.base/jdk.internal.util.Preconditions.checkFromToIndex(Preconditions.java:349)
	at java.base/java.lang.String.checkBoundsBeginEnd(String.java:4608)
	at java.base/java.lang.String.substring(String.java:2720)
	at java.base/java.lang.String.substring(String.java:2693)
	at com.thecoderscorner.tcmenu.javaapi/com.thecoderscorner.menu.persist.LocaleMappingHandler.getFromLocaleWithDefault(LocaleMappingHandler.java:23)
	at com.thecoderscorner.tcmenu.javaapi/com.thecoderscorner.menu.domain.util.MenuItemFormatter.bundleIfPossible(MenuItemFormatter.java:31)
	at com.thecoderscorner.tcmenu.javaapi/com.thecoderscorner.menu.domain.util.MenuItemFormatter.formatAnalogForDisplay(MenuItemFormatter.java:203)
	at com.thecoderscorner.tcmenu.javaapi/com.thecoderscorner.menu.domain.util.MenuItemFormatter.formatForDisplay(MenuItemFormatter.java:145)
	at com.thecoderscorner.tcmenu.menuEditorUI/com.thecoderscorner.menu.editorui.uimodel.UIAnalogMenuItem.populateMinMaxLabel(UIAnalogMenuItem.java:207)
	at com.thecoderscorner.tcmenu.menuEditorUI/com.thecoderscorner.menu.editorui.uimodel.UIAnalogMenuItem.internalInitPanel(UIAnalogMenuItem.java:110)
	at com.thecoderscorner.tcmenu.menuEditorUI/com.thecoderscorner.menu.editorui.uimodel.UIMenuItem.initPanel(UIMenuItem.java:224)
	at com.thecoderscorner.tcmenu.menuEditorUI/com.thecoderscorner.menu.editorui.controller.MenuEditorController.lambda$onTreeChangeSelection$10(MenuEditorController.java:271)
	at java.base/java.util.Optional.ifPresentOrElse(Optional.java:196)
	at com.thecoderscorner.tcmenu.menuEditorUI/com.thecoderscorner.menu.editorui.controller.MenuEditorController.onTreeChangeSelection(MenuEditorController.java:270)
[... 54 lines cut ...]
```

**Expected behavior**
A clear and concise description of what you expected to happen.

I can edit the Analog menu item in the designer and there is no exception in the logs.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
> Note that the right pane is still on the `BatteryVoltage` despite another item was clicked on the left.

**Desktop UI issue (please complete the following information):**
 - OS: All; reproduced on Windows 11 and Linux
 - Version Win11, Ubutu 23.04, Arch Linux
 - Output of diagnostics from tcMenuDesigner - not needed. See traces above.
 - Designer Version and any other relevant details - experimental beta

**Embedded environment
 Provide as much information as possible about the board configuration here.
- N/A

**Additional context**
Add any other context about the problem here.
- N/A

davetcc, 2023-05-01:
Yep, I noticed this myself just the other day. I think I worked out what's wrong but haven't had a chance to test it properly yet.

davetcc, 2023-05-02:
I've had a go at this after a code walkthrough I noticed a couple of errors, it may be worth you looking through the commit for this issue to see if it looks right

vzahradnik, 2023-05-02:
I'll check tomorrow. Thanks!

vzahradnik, 2023-05-04:
@davetcc I confirmed your fix solves this problem. I found another but I'll create a separate issue for it.

davetcc, 2023-05-07:
I found the underlying bug with this, and reopened it to do the commit.

davetcc, 2023-05-07:
Done, this item should be fully fixed now

closed 2023-05-07 by davetcc