TcMenu/tcMenu #353: Code generator: Java projectEmbedded EEPROM error

vzahradnik, 2023-05-13:
**Describe the bug**
[image]

I have an existing TcMenu project for ESP32. I copied the file as-is to generate Java server with the same functionality to run locally on my PC. The ESP32 project uses EEPROM class implementation. While the Designer UI says that Java will use inbuilt storage, when I generate the code the log shows an error.

I managed to generate the code by first setting target platform back to ESP32, setting EEPROM implementation to No EEPROM and then changing target to RPi Java.

**Commerical or Personal use**
Commercial use / Lutemi.

**To Reproduce**
- Download the following project file: https://gist.github.com/vzahradnik/8ddb94ffc710a0212860128c6812888a
- Open in TcMenu Designer
- Change Embedded platform to Raspberry Pi Java
- Generate the code

**Expected behavior**
No error in the generator. Settings will be stored using inbuilt storage as is written in the UI.

**Embedded environment**
ESP32 / Raspberry Pi Java. Only code generation issue.

davetcc, 2023-05-14:
Good spot there, I'd only ever tried creating a RPI project from scratch. Either code generator should ignore the EEPROM, or it should get blanked out when Java is selected. The later is a bit dangerous if someone selects it by accident and loses all ROM settings.

davetcc, 2023-05-14:
I had a go at fixing this today by just removing the exception, I didn't really see what the exception was trying to achieve. This code is quite well tested and all the tests still pass afterward meaning it didn't seem to achieve much. Can you try it again when you get time?

vzahradnik, 2023-05-14:
I'll check tomorrow.

vzahradnik, 2023-05-16:
It works fine. Closing this issue.