TcMenu/tcMenu #354: The step divisibility issue in analogmenuitem

prowong, 2023-05-15:
**Describe the bug**
There are 256 numbers from 0 to 255, it should be divisible by 16, so I think it should be judged whether the sum of Maximum value plus Offset value plus 1 can be divisible by step value.

[image]

prowong, 2023-05-15:
Offset from zero: 0
Max value:255
so the setting range is from 0 to 255, 256 numbers, if the step value is 16,  
(0+255+1)/16=16
it is divisible. If we follow the prompt and set the maximum value to 256, then the actual value range will be 0~256, not the expected 0~255.

davetcc, 2023-05-19:
I think based on the original spec the behavior is correct. The original issue was raised by @vzahradnik and I'll ask if it is working as per orginal issue?

IIRC It was more designed for smaller step sizes like 2, to allow for even only iteration for example.

For what you want to do with such large steps, you could you a scroll choice item without too much trouble.

vzahradnik, 2023-05-20:
I personally don't find any issue with the current implementation, at least for my needs. @davetcc can you link the original issue I raised here? I'm not sure what it was about. Thanks!

davetcc, 2023-05-20:
https://github.com/davetcc/tcMenu/issues/221 is the original issue..

davetcc, 2023-05-22:
Yep, I agree I think the implementation is correct too, it is designed for small steps such as 2 or 4, and assumes that you'd always in that case want even numbers. IE 0,2,4,8 etc. If you wanted some kind of custom arrangement use a scroll choice menu item in custom mode @prowong. Let me know if either of you disagree here, but otherwise I'll close in a day or two.

prowong, 2023-05-23:
Maybe I didn't describe it clearly enough. In fact the generated code works, I mean the prompt logic of the step limit is a bit of a problem, it should not just be based on whether the maximum value is divisible by the step size. My situation is similar to the original question. For example, the contrast setting range of the u8g2 library is 0-255. I want to set it to 16 levels of adjustment. There are 256 numbers from 0 to 255, that is, there are 16 steps, right? However, if you fill in 16 in the designer, it will prompt that the maximum value of 255 is invalid. If you fill in the maximum value of 256, it will exceed the range accepted by u8g2.  Is this something wrong with my understanding?

prowong, 2023-05-23:
This is my confusion, although there are 256 numbers from 0 to 255, but only 255 steps can be adjusted. I am sorry to trouble you.

closed 2023-05-23 by prowong

davetcc, 2023-05-23:
Yep it seems strange but if you do the math yourself it works out as 256. One easy option is to either use the max function to limit to 255 or use a scroll choice or enum o
Item for the steps