TcMenu/tcMenu #362: using DarkModeTraditional with eTFT and seperated generated source files creates invalid include

emaayan, 2023-06-04:
**To Reproduce**
select "Project in src folder  / seperate generated fodler " 
select  TFT_eSPI display 
select Dark Mode tradtional for small to medium color dispalys 
Generate coce 

**Expected behavior**
files shoudld be able to be compiled

**Actual**
file xx _menu.h is created with this include (shjould not be using brackets but " ) 
 
`#include <../ThemeDarkModeTraditional.h>`

additonally it  seems other themes create this in cpp file and not in header

AtmosphEng, 2023-06-05:
FYI:  I've experienced something similar with tcmenu 3.1.0 for a while but didn't get around to reporting it until now.  When I use the Designer Code Generation with the ThemeDarkModeTraditional on PlatformIO, I get the following linking error:

Linking .pio\build\lilygo-t-embed\firmware.elf
c:/users/david/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/lilygo-t-embed/src/TcmSimpleTest_menu.cpp.o: in function `installDarkModeTraditionalTheme(tcgfx::GraphicsDeviceRenderer&, tcgfx::MenuFontDef const&, tcgfx::MenuFontDef const&, bool)':
D:\SBOX\WORK\aaaPROJ\aaaProjects\TcmSimpleTest/src/ThemeDarkModeTraditional.h:16: multiple definition of `installDarkModeTraditionalTheme(tcgfx::GraphicsDeviceRenderer&, tcgfx::MenuFontDef const&, tcgfx::MenuFontDef const&, bool)'; .pio/build/lilygo-t-embed/src/TcmSimpleTest_main.cpp.o:D:\SBOX\WORK\aaaPROJ\aaaProjects\TcmSimpleTest/src/ThemeDarkModeTraditional.h:16: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\lilygo-t-embed\firmware.elf] Error 1
===================================================== [FAILED] Took 8.39 seconds ===

My workaround (found by experiment) has been to edit xyz_menu.h and cut-out the line containing: 
#include <ThemeDarkModeTraditional.h>, and instead  paste it at the top of xyz_menu.cpp. It then compiles OK. However, I need to do this cut and paste procedure after every Code Generation.

davetcc, 2023-06-06:
Thanks for reporting this, I'll go through the theme plugins and check them all.

closed 2023-07-26 by davetcc