TcMenu/tcMenu #371: Option to save settings to EEPROM after menu item change

vzahradnik, 2023-07-01:
**Is your feature request related to a problem? Please describe.**
TcMenu approach to saving settings is either:
- Save settings periodically. This is discouraged in the documentation due to EEPROM teardown
- Save settings when the [voltage drops](https://www.thecoderscorner.com/electronics/microcontrollers/psu-control/detecting-power-loss-in-powersupply/) below certain threshold.

I work with a custom board where I don't have any circuit to detect voltage drop and to react to the condition. At the moment, I save settings once in 5 minutes but this seems excessive considering users would rarely change the settings.

**Describe the solution you'd like**
TcMenu could add a new option to automatically save settings when any of the settings values is changed. Because users sometimes change text fields, there should be a callback once the text editing mode has ended.

Further improvement: only the EEPROM address for the changed field is overwritten. TcMenu already has support for this.

It is possible to implement this feature per app but I think that direct library support would be better.

**Additional context**
N/A

davetcc, 2023-07-05:
There is already a commit handler/menu manager observer that allows for capturing the end of an edit, it could be hooked onto that for anything that has a none -1 EEPROM location. 

This is probably less than 10 lines of code to be honest including the include statement, the problem with building it in is that people think it is always a good thing to do, but it risks damaging the EEPROM if there are many writes to a field. Then what about external writes from the remote API, they are not captured by this commit handler.

The only certain way would be to do a save on change but do it a maximum of every 10 minutes or something like that, and only when there is an actual change. The remote API change side needs a bit more thought, maybe there needs to be some notification from the incoming message processor side.

vzahradnik, 2023-07-10:
OK, thanks for clarification. I will go with the observer for now. You're right that damaging the EEPROM poses a risk.

davetcc, 2023-07-11:
The problem is, that I don't want to put something even remotely dangerous in the library itself and have a load of people turn up in the discussions six months after releasing their product with broken boards.

davetcc, 2023-07-11:
The best option is power-down detection, the second best option is nearly always to either use non-volatile RAM or a timed write to EEPROM that is delayed enough so that even if it wrote every time it would not affect the lifespan of the product. But even worst case at 10 minutes and 100,000 writes that gives you 1,000,000 minutes, or another way - that gives you 6,944 days, (or 19 years). I'd say that was probably a reasonable worst-case compromise.

vzahradnik, 2023-07-17:
Thanks for the hints. After some discussion with colleagues I decided to implement an option "Save settings" in TcMenu. It is simple solution and because people won't change settings often - usually just once during setup - this should work great.

Closing the issue.

closed 2023-07-17 by vzahradnik