TcMenu/tcMenu #374: [i18n] Display field values instead of cryptic %menu.[number].name

vzahradnik, 2023-07-04:
**Is your feature request related to a problem? Please describe.**
After the internationalization is enabled, the fields in the EEPROM layout window are not readable. Instead, users see the reference to the text field, like %menu.35.name.

**Describe the solution you'd like**
Technically it's not a bug but it would be great to load at least strings from the default language. Seeing layout like this makes the layout view not as usable.
[image]

davetcc, 2023-07-05:
Agreed, will add to 4.1

davetcc, 2023-07-22:
Also found a few more places, fixed them up at the same time:

* On the analog item page, the example values shown in labels were also wrong, showing units that were not localized.
* On the search menu item page, there were also similar issues, fixed there too.

davetcc, 2023-07-22:
Same for the "Menu Item -> Show Preview" option, now correctly uses the currently selected locale as selected when opened.

closed 2023-07-26 by davetcc