TcMenu/tcMenu #378: Add support for ESP32 Preferences library

vzahradnik, 2023-07-17:
**Is your feature request related to a problem? Please describe.**
ESP32 deprecated EEPROM approach in favor of Preferences library. More information and examples here:
https://randomnerdtutorials.com/esp32-save-data-permanently-preferences/

Also, this is discussed in TcMenu thread here:
https://github.com/davetcc/tcMenu/discussions/359

**Describe the solution you'd like**
Add Preferences support to TcMenu designer as another option next to EEPROM.

**Additional context**
N/A

tomelgato, 2024-04-12:
Beside the option for Preferences.h it would be nice to be able to enter a namespace which will be used for tcmenu

EEprom support : Preferences.h
Namespace: "......."

davetcc, 2026-01-17:
TcMenu designer integration will be released in 4.5, already releaesed in Ioabstraction

closed 2026-01-17 by davetcc