TcMenu/tcMenu #380: Improve the way choosing locales works

davetcc, 2023-07-22:
The way locale choice works is not right, it is buried deep within the name editor field within a menu item, where really it affects every part of the UI once changed. The "item is not localized" option is tied in with this and that is really wrong.

**Describe the solution you'd like**
Untie the item is localized from the locale choice, so that locale choice is done at the main window (above the tree seems most logical), and the ability to have menu items that are not localized is done by a toggle button on the menu item itself. This works much more safely.

**Additional context**
Choosing the toggle on entry is done by checking if the item starts with `%` and does not start with `%%`, this means it is not localized, and the box will be ticked in this case, otherwise it will be unticked, and any changes saved to the selected locale.

davetcc, 2023-07-22:
@vzahradnik I'd be interested in your feedback for this (if you had time to build it and see what you think), with this and the other fixes in #374 this should really improve the language support.

davetcc, 2023-07-22:
The language selection is at the top above the tree
[image]

If an item does not need to be localized as the name/unit/enum/list is universal then this is now indicated by this toggle button next to the name
[image]

vzahradnik, 2023-07-22:
Sure, I'll test and report back.

One thing that I noticed already by using this feature extensively is that TcMenu holds all strings in memory. Often when I include new strings by editing files, I have to exit TcMenu, start it again and load the project file. Otherwise the generated lang files don't contain new strings.

I don't know how difficult it would be to detect that a file has been modified but it would speed up the workflow. IDEs have this sorted out.

closed 2023-07-26 by davetcc

vzahradnik, 2023-07-27:
I just tried the latest master. Great work with this. It's much more usable than before.

I have one thing into discussion. Would it be difficult to mark certain attributes as untranslatable? I would like to prevent certain strings to be localized, for example Timezone offset like UTC+01:00.

In its current implementation, this feature is fully usable. I use it everyday now. Of course, I'll create GitHub issues once I bump or something.