TcMenu/tcMenu #389: TCMENU 4.0 UI bug

tomelgato, 2023-08-18:
There seems to be a little bug when you try to move a menu entry before or after a SubMenuItem. You can drag it on the SubMenuItem to move it inside but there is no way to put it before the Submenu. This becomes obvious when you have 2 SubMenuItems and you like to move something in between
(running on OS X)

[image]

vzahradnik, 2023-08-27:
I too have encountered issues like this. As far as I know this is not a new bug in TcMenu 4.1.1. Also moving items from submenu to the top hierarchy is buggy. Sometimes I edit the project file in text editor to achieve the hierarchy I want.

davetcc, 2023-08-28:
Can we please come up with a list of the cases that don't work properly and I will try and fix them in 4.2?

davetcc, 2023-08-28:
I'm going to have to do a patch release next day or two anyway as there is a critical bug that would affect new users around the create project workflow. It was not caught by the unit tests so managed to get into the release! I think I have a fix that makes this behaviour much better for at least the simple the cases.

closed 2023-08-29 by davetcc

tomelgato, 2023-09-05:
I can confirm, 4.1.2 fixes that issue! Thanks!