TcMenu/tcMenu #401: tcMenuDesigner-4.1.2.dmg broken

tomelgato, 2023-09-03:
The dmg seems to be corrupted

davetcc, 2023-09-03:
Thanks for letting me know, I’ll double check it myself later and rebuild if needed.

davetcc, 2023-09-03:
Most likely the notorization failed and then it says the image is damaged. Hard to detect that on development devices as they accept the developer certificates.

davetcc, 2023-09-03:
It's not that, it is properly notarized, and working on Intel Macs but not on a test Apple Silicon I tried, this may take a little time to work out what to do, as I need to work out why it failed first.

davetcc, 2023-09-03:
The good news is that I have an Apple Silicon to hand, so I can debug it and find out what is wrong.

davetcc, 2023-09-04:
I've got a good idea of what's going on now, after using an Apple Silicon Mac I recreated the issue. I think that Apple has stopped shipping Rosetta 2 by default, which means the Intel build will no longer work by default. Fortunately, we have an Apple Silicon Mac but it is not a developer build. 

What we plan to do for now is to install "xcode tools" and a "Java JDK 20" so that we can do the build from there, but this will take a bit of time as I need to get certificates onto that machine so it is provisioned for Notarization. We'll then generate two separate images - IE Intel and arm64.

tomelgato, 2023-09-04:
I downloaded it 3 times on High Sierra (Intel ;) ) .... it doesnt open by diskimage mounter and also a archive utility cannot do anything with it.

tomelgato, 2023-09-04:
I downloaded it again right now and its working. Maybe a hickup on my side!?

davetcc, 2023-09-05:
Can’t be a hiccup as I saw the same on a non development mac too.

 Something I read is that if the internet is not available at the time the image is loaded it can fail to notarise, but honestly at this point I don’t understand it, I’ll maybe first do a binary comparison of the source and uploaded file. Then raise with Apple if needed to get some guidance, the dev program comes with some support.

tomelgato, 2023-09-05:
Did you reupload the file during the last 2 days? If the file is stored in a distributed content delivery network maybe some nodes have a faulty file. Just a wild guess

davetcc, 2023-09-05:
I didn't reupload, but maybe there was a bad node that was taken down. That's all I can think, I will try it again later today on another Intel mac that is not provisioned with developer licenses. I think the Apple Silicon issue is separate, and I imagine the reason there are not a lot of reports on that, is because most who do embedded development have Rosseta 2 installed.

Never the less, it's probably about time to provide a native build for them, I'll start with 4.2