TcMenu/tcMenu #402: TcMenu 4.0 -> 4.1 not working

tomelgato, 2023-09-04:
Strange problem with 4.1.1:

|-- TaskManagerIO @ 1.4.1
|-- SimpleCollections @ 1.2.3
|-- tcMenu @ 4.1.1
|-- tcUnicodeHelper @ 1.0.0
|-- IoAbstraction @ 4.0.2
|-- LiquidCrystalIO @ 1.4.3

I switched now back and forth several times between 4.0 and 4.1.1. With 4.0 everything works, with 4.1.1 the menu displays but doesnt react

I see in my extended keyboard routine that it catches the keys and i send them to 

MenuEditingKeyListener::keyPressed(key,held);

but no reaction on the menu at all. Afair i tried yesterday night also the normal keyboardhandler with the same result.

What do you need to track down the problem?

P.S. i did regenerate the code with Tcmenu Designer 4.1.1

davetcc, 2023-09-05:
Can you provide the keyboard listener so I can have a look at it, feel free to email it to avoid public disclosure if need be. I'll delete it once I have taken a look. There was a one-off change over the 4.0 and 4.1 releases to make rendering more reliable and I suspect your custom keyboard has been hit by those changes.

davetcc, 2023-09-05:
BTW - We have two different keyboard-based menus in our test pack, one on an ATMEGA2560 AVR with LCD, and another on a Seeed MG126 SAMD with TFT. One uses directly wired keys, the other uses keys wired on an MCP23017. Both tests passed with 4.1 before release.

tomelgato, 2023-09-05:
I can send you my emf and the generated code files. My keyboard listener code you see there: https://github.com/davetcc/tcMenu/discussions/309
Btw. what happens when you have a non valid value in the eeprom for a given menu item?

tomelgato, 2023-09-05:
Beside the non working navigation tcmenu works. I get the keys via MenuEditingKeyListener, when i change a menu item with setCurrentValue i see the change on the display

tomelgato, 2023-09-05:
Like said, i dont think its the listener. Afair i disabled it during my error tracking and used the normal way too. Can try it again. Need to do a testing branch

tomelgato, 2023-09-05:
Ok i tried again with TCMenu Designer 4.1.2 and the original calls:

MatrixKeyboardManager keyboard;
MenuEditingKeyListener menuKeyListener;

...

keyboard.initialise(ioUsingArduino(), &keyLayout, &menuKeyListener);

Same as with my extend call

davetcc, 2023-09-08:
The problem here is that the examples with keyboards work so I have no reference point to work from, what I would suggest is to try making minimal modifications to the dynamicMenuItems example to meet your board spec and see if that works.

As I said previously, given this is caused by me changing the rendering system to fix most of the outstanding issues, I would as a one-off be happy to look over the code if you zip up the project and email it to me to avoid public disclosure. I'll compare it to what I have working basically.

tomelgato, 2023-09-08:
I do a mini test project for you with pleasure! Lets see if the bug persists after stripping down my project

tomelgato, 2023-09-13:
@davetcc I did send you the project to your email

tomelgato, 2023-10-04:
Ping

davetcc, 2023-10-20:
As per the email, can you feedback on the example, please?

tomelgato, 2023-10-29:
Thanks, did answer your mail

davetcc, 2024-01-14:
I think I've _finally_ found it. All my keyboard-based test harnesses had rotary encoders on them as well..

I just created a new board without the encoder, boom, I could not see the edit and active item cursor, and could not change position. It was caused because there has to be an encoder (even if it is an empty one), I'll be making a commit against this hopefully today and it should therefore be fixed in 4.2. If the change is as simple as I expect, you could try it easily once I commit the change.

tomelgato, 2024-01-15:
Holy thread revival. Will try it out!

davetcc, 2024-01-15:
Committed it just now. It worked locally without a rotary encoder for me on
a fairly large menu.

davetcc, 2024-01-15:
The only real change (that will need a change in the plugins to work) was adding this method to the keyboard listener in the library:

      tcMenuKeyListener.createInternalEncoder();

This makes sure there is a default encoder. If there is not one, it creates it. You could simply try for now and add:

    switches.setEncoder(0, new RotaryEncoder([](int v) {
        menuMgr.valueChanged(v);
    }));

after the setupMenu() call.

closed 2024-01-16 by davetcc

tomelgato, 2024-02-25:
Hello Dave!

Your hack works with tcmenu 4.2.1

Without it still doesnt work with this version, not sure if you said you want to intgrate in 4.2.x

Cheers - Tom