TcMenu/tcMenu #415: [TcMenuDesigner] Add an settings option to set font size and optionally other UI settings

vzahradnik, 2023-11-06:
**Is your feature request related to a problem? Please describe.**
I would like to change the default font size to perform a live demo in front of an audience. Currently TcMenu doesn't reflect system font scaling settings nor it has an settings option to configure font size. As the default font size is quite small, it would be useful for other usecases. However, this will affect all existing dialogs which already have problem displaying longer texts.

**Describe the solution you'd like**
User can increase/decrease font size used in menus.

**Additional context**
N/A

davetcc, 2023-11-06:
Agreed and should be a simple change. 4.2 is now just around the corner and if it is as easy as this without too many side effects it may be possible. As you say some dialog already struggle with longer text. I’m considering removing much of the fixed sizing from fxml at some point soon to see if things improve. See link 

https://stackoverflow.com/questions/32999605/how-can-i-change-the-base-font-size-in-javafx-8-modena-theme-that-would-cascade

davetcc, 2023-11-06:
I’ve assigned it to me but feel free to have a go at that suggested fix. There is a css file in the tcMenuGenerator resources dir.

davetcc, 2023-12-10:
I've been doing this in the background, it is mainly just removing direct sizes put in by the form designer, and replacing them with layouts instead.

davetcc, 2023-12-10:
it is more reliable and easier than expected, I'm about 70% through.

closed 2023-12-19 by davetcc