TcMenu/tcMenu #416: TcMenu 4.1.2 not released in PlatformIO

vzahradnik, 2023-11-07:
**Describe the bug**
Latest TcMenu, version 4.1.2, is not published in PlatformIO registry.

**To Reproduce**
Check the project page here:
https://registry.platformio.org/libraries/davetcc/tcMenu/installation

**Expected behavior**
Version 4.1.2 is available to install.

davetcc, 2023-11-07:
This will actually be in the library repos as the Arduino spec requires one library per repository. See tcMenuLib repo for the code

vzahradnik, 2023-11-07:
Sorry, you're right. I created this issue rather quickly.

davetcc, 2023-11-07:
No problem I usually create a library tracking issue anyway. For visibility here.

davetcc, 2023-11-08:
4.1.2 was not released even to Arduino, I was going to release as 4.1.2 but decided to "round up" into 4.2

vzahradnik, 2024-09-08:
After the ownership of the TcMenu packages in the PlatformIO registry changes to the TcMenu organization, we will create a new deploy target to be triggered on library release.

closed 2026-09-06 by davetcc