TcMenu/tcMenu #417: Cannot open any project

leelx78, 2023-11-10:
Hi @davetcc just noticed that on Mac M1 OS 14.1.1 I can no longer open any project. It's like any window other than the main is blocked.

To reproduce, just open TC Menu Designer v 4.1.2 go to File->Open and then nothing happens. Same with File->Save Project As
and so forth.

I had a look on the Console and this is what I see:
```
error	16:15:57.589430+0000	tcMenuDesigner	Java Exception
error	16:15:57.590523+0000	tcMenuDesigner	(
	0   CoreFoundation                      0x00007ff810841aae __exceptionPreprocess + 242
	1   libobjc.A.dylib                     0x00007ff8103385ff objc_exception_throw + 48
	2   CoreFoundation                      0x00007ff81084194e +[NSException raise:format:] + 206
	3   CoreFoundation                      0x00007ff8107c4ef6 __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 137
	4   CoreFoundation                      0x00007ff810854cac ___CFXRegistrationPost_block_invoke + 86
	5   CoreFoundation                      0x00007ff810854bfd _CFXRegistrationPost + 530
	6   CoreFoundation                      0x00007ff810794e49 _CFXNotificationPost + 826
	7   Foundation                          0x00007ff81167622e -[NSNotificationCenter postNotificationName:object:userInfo:] + 82
	8   AppKit                              0x00007ff813ed464b -[NSApplication _handleDeactivateEvent:] + 925
	9   AppKit                              0x00007ff814631d9d -[NSApplication(NSEventRouting) sendEvent:] + 2305
	10  AppKit                              0x00007ff8141f20df -[NSApplication _handleEvent:] + 65
	11  AppKit                              0x00007ff813d30e1f -[NSApplication run] + 640
	12  libglass.dylib                      0x0000000135241037 -[GlassApplication runLoop:] + 1895
	13  Foundation                          0x00007ff8116e9ca5 __NSThreadPerformPerform + 178
	14  CoreFoundation                      0x00007ff8107cfa16 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
	15  CoreFoundation                      0x00007ff8107cf9b9 __CFRunLoopDoSource0 + 157
	16  CoreFoundation                      0x00007ff8107cf788 __CFRunLoopDoSources0 + 215
	17  CoreFoundation                      0x00007ff8107ce3f8 __CFRunLoopRun + 919
	18  CoreFoundation                      0x00007ff8107cda99 CFRunLoopRunSpecific + 557
	19  libjli.dylib                        0x000000010901dba2 CreateExecutionEnvironment + 402
	20  libjli.dylib                        0x00000001090196a7 JLI_Launch + 1543
	21  tcMenuDesigner                      0x0000000100a8257a jvmLauncherStartJvm + 298
	22  tcMenuDesigner                      0x0000000100a80b22 _ZN3Jvm6launchEv + 850
	23  tcMenuDesigner                      0x0000000100a840f1 _ZN12_GLOBAL__N_115initJvmLauncherEv + 2337
	24  tcMenuDesigner                      0x0000000100a87c51 _ZN3app6launchERKSt9nothrow_tPFvvEP11LogAppender + 257
	25  dyld                                0x0000000200c003a6 start + 1942
)
error	16:15:57.649899+0000	ContextStoreAgent	Simulating crash. Reason: <private>
error	16:15:57.650868+0000	ContextStoreAgent	Simulating crash. Reason: <private>
error	16:15:57.660726+0000	WindowServer	Clearing datagram buffer for cid 0x7a793. Client state: Data count: 16384 connectionIsUnresponsive: 0 clientMayIgnoreEvents: 1 clientIsSuspended: 0

```
Can you reproduce at all? Hopefully Apple didn't break it with the last update!

leelx78, 2023-11-10:
this my java:

```
➜  ~ java --version
openjdk 11.0.15 2022-04-19
OpenJDK Runtime Environment (build 11.0.15+0-b2043.56-8887301)
OpenJDK 64-Bit Server VM (build 11.0.15+0-b2043.56-8887301, mixed mode)
```

davetcc, 2023-11-10:
We have many people using this version on MacOS including me! Did you upgrade to the new version?

Honestly I really don’t like large MacOS versions, they invariably break one of my dev environments. I try and hold off literally as long as I can. 

unless you’ve built it yourself it will be using the packaged executable, and that is based on Liberica 20. Presently, we’re relying on Rosetta to translate the x64 image. But this time around for 4.2 I’ll try build a native image.

Can you check the logs in .tcmenu directory to see what’s in there too?

davetcc, 2023-11-10:
Unfortunately it is a JFX UI framework bug that presently doesn’t have a fix. https://bugs.openjdk.org/browse/JDK-8318129?page=com.atlassian.jira.plugin.system.issuetabpanels%3Achangehistory-tabpanel

davetcc, 2023-11-11:
If you are using tcMenu Designer (or any other Java-based applications on the Mac), **do not upgrade to macOS 14 at the moment**, there are currently major issues. How such a severe interaction in Java escaped Apple's testing is beyond me.

davetcc, 2023-11-11:
As soon as that fix makes it's way into an OpenJDK build, I'll rebuild the macOS package against it, even out of cycle.

leelx78, 2023-11-11:
Hi @davetcc unfortunately I need to keep my machine up-to-date with the latest OS and apps by contract so I am running the very latest "everything" from Apple.

The issue only seems to affect opening new windows but the rest seems fine. Code generation also works.

Any other possible way around it? I'll see if I can run it in a virtual machine in Parallels but it's not ideal..

davetcc, 2023-11-11:
Open menus from the CLI until it is fixed.  See the documentation on the CLI.

davetcc, 2023-11-11:
For CLI see https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/tcmenu-cli-workflow/.

Basically, first setup terminal to path in the tcmenu command as in the link above, then use `tcmenu gui` in the directory with the emf, it will start with that project loaded. 

I think there is already a JDK 21 critical patch, but I don't think it includes a fix for this, I need to check and work out if it is fixed yet.

leelx78, 2023-11-11:
Hi for info I just installed JDK21 via homebrew
```
➜  ~ $(brew --prefix openjdk)/bin/java --version
openjdk 21.0.1 2023-10-17
OpenJDK Runtime Environment Homebrew (build 21.0.1)
OpenJDK 64-Bit Server VM Homebrew (build 21.0.1, mixed mode, sharing)
➜  ~
➜  ~
➜  ~
➜  ~ /usr/libexec/java_home -V
Matching Java Virtual Machines (3):
    21 (arm64) "Oracle Corporation" - "Java SE 21" /Library/Java/JavaVirtualMachines/jdk-21.jdk/Contents/Home
    1.8.351.10 (x86_64) "Oracle Corporation" - "Java" /Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home
    1.8.0_362 (arm64) "Amazon" - "Amazon Corretto 8" /Users/lele/Library/Java/JavaVirtualMachines/corretto-1.8.0_362/Contents/Home
/Library/Java/JavaVirtualMachines/jdk-21.jdk/Contents/Home
➜  ~
```

and launched tcmenudesigner it again. The dialog still doesn't open but I didn't see an error in the console but I see this:
```
default	11:31:36.027267+0000	tcMenuDesigner	window NSPopupMenuWindow c31 finishing close
default	11:31:36.028490+0000	tcMenuDesigner	order window: c31 op: 0 relative: 0 related: 0
default	11:31:36.030433+0000	tcMenuDesigner	[0x600003a640f0] activating connection: mach=false listener=false peer=false name=com.apple.TextInputUI.xpc.CursorUIViewService
default	11:31:36.031083+0000	tcMenuDesigner	[0x600003a640f0] invalidated after the last release of the connection object
default	11:31:36.031299+0000	tcMenuDesigner	[0x7fab7d5090e0] activating connection: mach=false listener=false peer=false name=(anonymous)
default	11:31:36.030795+0000	CursorUIViewService	[0x12360f5e0] activating connection: mach=false listener=false peer=true name=com.apple.TextInputUI.xpc.CursorUIViewService.peer.0x12360f5e0
default	11:31:36.031430+0000	tcMenuDesigner	[0x7fab7d50a090] activating connection: mach=false listener=false peer=false name=(anonymous)
```

Can I be sure tcMenuDesigner is using this JDK or do I need to set it up separately?

davetcc, 2023-11-11:
Your JDK is not used. Since several years now it is self contained and jlinked to an executable. You’d need to build yourself to use that JDK.

davetcc, 2023-11-11:
The thing is I don't think it is fixed by that JDK as I don't see it in the release list, I'll download it later or tomorrow and do a build that you can try as it doesn't take long to do a build. In the mean-time using the CLI to open a new project should work.

davetcc, 2023-11-11:
Basically, anything that uses the desktop integration (IE file open/save, browse link etc) will fail.

leelx78, 2023-11-11:
ok thanks @davetcc. for the time being the CLI is ok for me. I'll try to sort out the encoder now.

davetcc, 2023-11-22:
have upgraded locally to JDK21, will produce a mac beta build soon.

davetcc, 2023-11-24:
Slight problem, I have a beta build ready, but Apple has changed the notarization process recently and the new process is not working for the app at the moment, I'm trying to work out what options I need to add to get it working again.

davetcc, 2023-11-25:
A BETA has been released that should fix the issue, I have upgraded my Intel based mac to Sonoma now and it is tested as working for me.

davetcc, 2023-11-25:
See the releases page, please close if it fixes the issue for you

leelx78, 2023-11-25:
Hi @davetcc  yes seems to work! Great job thanks a lot!!

closed 2023-11-25 by leelx78