TcMenu/tcMenu #419: i18n feature is buggy

vzahradnik, 2023-11-17:
**Describe the bug**
i18n language support in designer is buggy. For example, I create a new text menu item, give it a name, then I hit on Save, and the name changes to "New variable". There is more...

**To Reproduce**
Activate i18n support by adding at least one extra language, try to modify the menu items.

davetcc, 2023-11-22:
I'll be creating a beta in a few days, it contains quite a few fixes along with the file watcher support. Let's keep this open until the i18n flow is at least fully workable.

vzahradnik, 2023-11-29:
Latest branch behaves better but I still found at least one issue.

To reproduce:
1. Add a new submenu item
2. Change the submenu item text
3. Save the project

=> Submenu item text should be discarded and changed to New SubMenu.

Attaching a video.

https://github.com/davetcc/tcMenu/assets/4460766/93ba14ed-5529-4c8e-bc76-2971bc143779

davetcc, 2023-12-21:
I'm going to close this for now, I believe most of the issues to be resolved by the simplification project, but if you find anything else, raise a new issue for the specific issue.

closed 2023-12-21 by davetcc