TcMenu/tcMenu #431: Encoder is not responsive with long tasks in loop

leelx78, 2023-12-08:
Hi @davetcc I open a card here because it looks like the issue is with the encoder class.

let's consider a case where there is a function in the main loop which takes a set amount of time (5.5ms in my case), e.g.

```
void loop()
{
  taskManager.runLoop();
 
  DEVICE.loopFunction(); // this function takes 5.5ms
}
```

In this case if I use a third party library for the encoder I can simply flag a "ENCODER_ROTATED_CW" into an ISR whenever the encoder transition happens and then react to it at the next loop. Works well but unfortunately if I use this approach I would need to manually pass these events into MenuManager which is not ideal.

If I use your rotary encoder implementation (either polling or via interrupts) it seems to miss most of the events. From what I see if the transition happens during DEVICE.loopFunction() it is just ignored, while the other library correctly flags it down.

I had a look at your RotaryEncoder classes but I couldn't see the issue so far. Do you have any suggestions? 

Thanks as always!

davetcc, 2023-12-11:
So in short, go with the inconvenience for now, knowing that a proper solution is not far away. The rotary encoder in IoAbstraction has been a bugbear of mine for a while, I need to improve it for a couple of situations myself. Once the new state machine encoder is working, it should be near flawless.

davetcc, 2023-12-19:
> and works great so maybe you don't need to reinvent the wheel and just include it in your work? The ISR does nothing but flagging an event (CW/CCW/SW) which is then addressed in the main loop at the appropriate time.

Which version of IoAbstraction are you using out of interest? 

After 4.2 goes live, I'm going to finish off the state-machine based encoder logic. The thing is TcMenu supports a few other environments other than Arduino, so I'll get the encoder support working for this case.

leelx78, 2023-12-19:
Morning @davetcc I'm using your latest available on platformIO
```
davetcc/tcMenu@^4.1.1
```

looking forward for the encoder support! Hopefully the library I mentioned gave you some feedback on what is working well on ESP32 S3

Thanks!

leelx78, 2023-12-21:
Hi @davetcc do you have an updated ETA for embedding the encoder class? My hack is sort-of-working but it's really not a good solution for production. I'll be more than happy to test a beta and give feedback if you have anything ready now.

Thanks!

vzahradnik, 2024-05-06:
@davetcc what is the current state of this issue? Is the state machine-based encoder still in work in progress? I have some issue with the rotary encoder myself and I wonder whether I should report bugs and/or try to fix the code.

glebv, 2024-08-14:
Hi, I have the same issue, could you provide a link to workaround?

davetcc, 2026-09-05:
This is probably one of the most tricky issues we've had to address. But I think we've now managed it.

There are now three implementations of encoder.

1. If you're prepared to either make A and B both interrupt driven, or provide an extra pin that is effectively hardware A or B, or use a hardware timer (such as SYS_TICK on ARM) then the new interrupt driven encoder is immune to long running tasks.
2. If you don't want to use interrupts/timers as above, and the encoder is on internal GPIO, then a middle ground fairly tolerant encoder will be used.
3. If the rotary encoder is on I2C, then the existing encoder will be used, there's no way to improve on that case.

davetcc, 2026-09-05:
https://github.com/TcMenu/IoAbstraction/issues/223

davetcc, 2026-09-05:
Already tested on PicoSDK. Will test on STM32Cube, SAMD, AVR, ESP32 boards.

closed 2026-09-05 by davetcc