TcMenu/tcMenu #441: TcMenu Designer did not start

TechCranky, 2024-01-07:
**Error  when launch**

"Could not load designer"
App did not start due to Cannot invoke
"String.startsWith(String)• because "<parameterl > " is null. See
log for more details.

[image]

**log report is given below:**

2024-01-07 14:03:48 INFO com.thecoderscorner.embedcontrol.core.util.TccDatabaseUtilities ensureTableFormatCorrect Checking for table TC_CONNECTION
2024-01-07 14:03:48 INFO com.thecoderscorner.embedcontrol.core.util.TccDatabaseUtilities ensureTableFormatCorrect Checking format of TC_CONNECTION
2024-01-07 14:03:48 INFO com.thecoderscorner.embedcontrol.core.util.TccDatabaseUtilities ensureTableFormatCorrect Table - TC_CONNECTION, changed= false
2024-01-07 14:03:48 INFO com.thecoderscorner.embedcontrol.core.util.TccDatabaseUtilities ensureTableFormatCorrect Checking for table TC_FORM
2024-01-07 14:03:48 INFO com.thecoderscorner.embedcontrol.core.util.TccDatabaseUtilities ensureTableFormatCorrect Checking format of TC_FORM
2024-01-07 14:03:48 INFO com.thecoderscorner.embedcontrol.core.util.TccDatabaseUtilities ensureTableFormatCorrect Table - TC_FORM, changed= false
2024-01-07 14:03:48 INFO com.thecoderscorner.embedcontrol.core.util.TccDatabaseUtilities ensureTableFormatCorrect Checking for table GLOBAL_SETTINGS
2024-01-07 14:03:48 INFO com.thecoderscorner.embedcontrol.core.util.TccDatabaseUtilities ensureTableFormatCorrect Checking format of GLOBAL_SETTINGS
2024-01-07 14:03:48 INFO com.thecoderscorner.embedcontrol.core.util.TccDatabaseUtilities ensureTableFormatCorrect Table - GLOBAL_SETTINGS, changed= false
2024-01-07 14:03:48 INFO com.thecoderscorner.embedcontrol.core.service.DatabaseAppDataStore getGlobalSettings Get global settings from store
2024-01-07 14:03:48 INFO com.thecoderscorner.embedcontrol.core.service.DatabaseAppDataStore getGlobalSettings Reading settings from table
2024-01-07 14:03:48 SEVERE com.thecoderscorner.menu.editorui.MenuEditorApp start Failed loading config
java.lang.NullPointerException: Cannot invoke "String.startsWith(String)" because "<parameter1>" is null
	at com.thecoderscorner.tcmenu.javaapi@4.1.2/com.thecoderscorner.menu.domain.state.PortableColor.<init>(Unknown Source)
	at com.thecoderscorner.embedcontrol.core/com.thecoderscorner.embedcontrol.core.service.TcPreferencesPersistence.populateGlobalSettings(Unknown Source)
	at com.thecoderscorner.embedcontrol.core/com.thecoderscorner.embedcontrol.core.service.CoreControlAppConfig.lambda$new$0(Unknown Source)
	at java.base/java.util.Optional.ifPresent(Unknown Source)
	at com.thecoderscorner.embedcontrol.core/com.thecoderscorner.embedcontrol.core.service.CoreControlAppConfig.<init>(Unknown Source)
	at com.thecoderscorner.tcmenu.menuEditorUI/com.thecoderscorner.menu.editorui.storage.MenuEditorConfig.<init>(Unknown Source)
	at com.thecoderscorner.tcmenu.menuEditorUI/com.thecoderscorner.menu.editorui.MenuEditorApp.start(Unknown Source)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.lambda$launchApplication1$9(Unknown Source)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runAndWait$12(Unknown Source)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runLater$10(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runLater$11(Unknown Source)
	at javafx.graphics/com.sun.glass.ui.InvokeLaterDispatcher$Future.run(Unknown Source)
	at javafx.graphics/com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at javafx.graphics/com.sun.glass.ui.win.WinApplication.lambda$runLoop$3(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)

davetcc, 2024-01-07:
Can you try backing up and removing the ~/.tcmenu directory and then start the app again, please?

It seems like something may have got into a bad state in the user preferences storage.

davetcc, 2024-01-27:
Closing as no response, please re-open if still an active issue.

closed 2024-01-27 by davetcc