TcMenu/tcMenu #453: Rebuild theme support on top of ThemeBuilder.

davetcc, 2024-02-29:
**Is your feature request related to a problem? Please describe.**
Themes have been a thorn in the side of designer for a long time, they expose far too much of the graphics plugin's internal workings and are exceedingly difficult to understand unless you've read the source code. This is obviously far from ideal, and a better solution is needed. Further, they are difficult to upgrade with the user needing to delete the file and then code generator rebuild, which again is far from optimal.

**Describe the solution you'd like**
Now that ThemeBuilder exists, it is possible to make the designer generate themes based on theme builder, which wraps all the complexity up and provides simple, easy to understand methods, which should be easier to deal with programmatically. Along with this if we provide a weakly implemented method (like the Arduino yield method), then we can allow users the option to extend the behaviour at build time.

Further, recent changes to create in memory bitmaps for the bitmap editor mean we can now even show a preview of the theme that looks like the real thing with only a small amount of work. So not only will the output be actually readable and updatable

**Additional context**
See the `TcThemeBuilder` class in the TcMenuLib library. See that some examples have already been upgraded, such as [PicoSDK OLED Encoder](https://github.com/davetcc/tccLibSdk/blob/master/examples/ControlAppOledEncoderPico/appTheme.cpp).

We are open to as much feedback as possible on this.

davetcc, 2024-07-28:
After thinking about this for a long time, IMHO there should be a couple of inbuilt "starter" themes that can be chosen in code generator, which get deployed into the theme header file (as now). Importantly, we should then document the theme builder support well enough that people can easily modify the provided template to meet their needs. @vzahradnik @Ales-Svoboda  what do you think?

vzahradnik, 2024-07-28:
I agree. Currently, the logic is so complex that even I am using just the default code generated by the TcMenu generator.

Ales-Svoboda, 2024-07-28:
I also agree.

closed 2024-09-01 by davetcc