TcMenu/tcMenu #455: No encoder, TCMenu 4.1+

tomelgato, 2024-03-07:
Refering to closed #402:

> I just created a new board without the encoder, boom, I could not see the
> edit and active item cursor, and could not change position. It was caused
> because there has to be an encoder (even if it is an empty one), I'll be
> making a commit against this hopefully today and it should therefore be
> fixed in 4.2. If the change is as simple as I expect, you could try it
> easily once I commit the change.

Seems it is still not working in 4.2. I still need this little hack which works:

> The only real change (that will need a change in the plugins to work) was adding this method to the keyboard listener in the library:
> 
>   tcMenuKeyListener.createInternalEncoder();
> This makes sure there is a default encoder. If there is not one, it creates it. You could simply try for now and add:
> 
> switches.setEncoder(0, new RotaryEncoder([](int v) {
>     menuMgr.valueChanged(v);
> }));
> after the setupMenu() call.

davetcc, 2024-03-09:
Let me look at this again to make sure there is an encoder in all cases, one possibility is a catch-all in the menu initialisation that ensures there is always an encoder present and creates one if not, then code elsewhere does not need to worry about it; IE the back stop to create it is there. I'll see if I can add it.

davetcc, 2024-03-09:
Tracking lib issue - https://github.com/davetcc/tcMenuLib/issues/213

davetcc, 2024-03-09:
could you try to patch your local tcMenu.cpp with the change in the above issue, it should create an encoder if there is not already one in the no input initialisation.

tomelgato, 2024-04-16:
Will test asap when im on that machine again

davetcc, 2024-07-28:
I'm going to close this for now, as I have personally tested keyboard support and know it to be working. Please reopen if this is still and issue along with an example application.

closed 2024-07-28 by davetcc