TcMenu/tcMenu #463: TCMenu 4.2.2 crashes on start

tomelgato, 2024-04-19:
Fresh downloaded and installed TCMenu 4.2.2 crashes on start. Is it compatible with 10.15?

Process:               tcMenuDesigner [44998]
Path:                  /Applications/tcMenuDesigner.app/Contents/MacOS/tcMenuDesigner
Identifier:            com.thecoderscorner.menu.editorui.cli
Version:               4.2.2 (4.2.2)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           tcMenuDesigner [44998]
User ID:               501

Date/Time:             2024-04-19 16:46:59.586 +0200
OS Version:            Mac OS X 10.15.7 (19H2026)
Report Version:        12
Anonymous UUID:        B332BC95-A7BF-31B8-02DE-1702C4484F69

Time Awake Since Boot: 240000 seconds

System Integrity Protection: enabled

Crashed Thread:        1

Exception Type:        EXC_BAD_ACCESS (SIGABRT)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00002b330d4040b0
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x2b330d4040b0:
    VM_ALLOCATE            00000008000e0000-0000000840000000 [  1.0G] ---/rwx SM=NUL  
--> 
    MALLOC_NANO            0000600000000000-0000600008000000 [128.0M] rw-/rwx SM=PRV  

Application Specific Information:
abort() called

davetcc, 2024-04-22:
Honestly, Apple is now by far the most difficult platform to support with problems after each MacOS release and with the notarization process too. On top of that, the general lack of funding means that it costs me money directly to keep a developer license.

Does it get far enough to start the JVM, or does it crash immediately? I have run the developer version very recently on MacOS. Let me get someone else to try and install it from the disk image on GitHub.

tomelgato, 2024-04-22:
4.1.2 runs on this machine under OS X 10.15.7, but not 4.2.2

4.2.2 runs on another machine under OSX 13.6.6

Where do i see how far it gets?

davetcc, 2024-04-22:
I would probably start with is anything at all in the log file in the `~.tcmenu` directory.

I don't think you'll find anything there, it looks like it is crashing very early on in the process.

Is this on an Intel or Apple Silicon mac? All the images are presently built from an Intel Mac.

davetcc, 2024-04-25:
We don't have any machines remotely near 10.15.7 Catalina to test it with as this is a 5-year-old operating system. In theory it _should_ work, but realities on the ground will most likely be different.

davetcc, 2024-04-25:
I recommend that you clear the log in the ~/.tcmenu directory, do a fresh run, see if anything gets logged, and then report back here if it even gets that far. Unfortunately, given the legacy nature of the OS you're using, it will be difficult to find others with a similar version. I can only recommend that you get whatever logging you can and post it here. If it is possible to apply a simple fix, I will put it in the next version. I don't like writing machines off, but I suspect it is a library compatibility issue caused by mandatory changes needed to work on the latest MacOS, and not at all fixable.

tomelgato, 2024-04-25:
Intel. Will update the machine to Ventura sooner or later.

tomelgato, 2024-04-26:
If its just Catalina related just update the requirements for TCMenu 4.2.2 onwards and leave it like this

davetcc, 2024-05-03:
Unfortunately, I suspect anything before 10.13 wouldn't work as that's the minimum version of xcode tools needed to build it.

However, I don't have enough resources to determine exactly where the cutoff is, but I imagine it 10.13.

closed 2024-05-03 by davetcc