TcMenu/tcMenu #465: addendum to the old update render thread

tomelgato, 2024-04-26:
Back then i was puzzled about the disappearing Root item (it was just there after start, when entering a submenu and going back it disappeared).

On a new project i found the cause:

renderer.setTitleRequired(false);    // now we turn off the title and change the editor characters  

This setting doesnt switch off the title completely, the title is there after start of tcmenu and disappears when you enter a menuitem (and doesnt reappear again)

davetcc, 2024-05-03:
You're supposed to call that method before the first time rendering takes place, or completely reset the rendering otherwise after calling it, as there is a cache in the renderer that holds the display order. 

See how the theme does it by refreshing the cache.

https://github.com/davetcc/tcMenu/blob/master/xmlPlugins/core-themes/coolBlue/ThemeCoolBlueModern.h#L40

tomelgato, 2024-05-03:
Oh yes, calling setupMenu() afterwards did do the trick

closed 2024-05-03 by tomelgato