TcMenu/tcMenu #468: Strange encoder behaviour when taken over display

tomelgato, 2024-05-02:
TCmenu 4.2

After taking over the display with
`renderer.takeOverDisplay(myDisplayCallback);`

it calls
```
void myDisplayCallback(unsigned int encodervalue, RenderPressMode clicked) {

}
```

but i just get encodervalue 0-3 and clicked stays on 1 (clicked) or 2 (held) till i turn the encoder again

do i miss something?

taskManager.runLoop(); is called frequently enough

tomelgato, 2024-05-02:
Can i poll the encoder directly when taken over the display? I would maybe need a quicker update frequency then 1/5 seconds.

davetcc, 2024-05-03:
> Can i poll the encoder directly when taken over the display? I would need a quicker update frequency then 1/5 seconds.

I've never tried it but I cannot see why not. You can even _I think_ check if the button is pressed using polling.

davetcc, 2024-05-03:
Can you please raise things like this as discussions, and not overwrite the issue template. You've done it several times now, those templates are there for a reason, to help us identify real problems from questions. It is hard enough to run a project of this size even when everyone follows the rules.

tomelgato, 2024-05-03:
Ok, no problem. I did see it as a bug, cause the callback is called over and over again with the same values. I expected to get just new values/states.

closed 2024-05-20 by tomelgato