TcMenu/tcMenu #470: Rotary Encoder plugin: Add a drop-down list with EncoderType

vzahradnik, 2024-05-06:
**Is your feature request related to a problem? Please describe.**
TcMenu defines the following enum for the rotary encoder type.
[image]

The plugin in TcMenu Designer, however, doesn't list all the options. `HALF_CYCLE` is missing and `FULL_CYCLE` is implicit.

[image]

**Describe the solution you'd like**
Instead of having a checkbox titled Quarter cycle encoder, add a new drop-down list with all encoder type options, defaulting to `FULL_CYCLE`.

davetcc, 2024-05-29:
Agreed that would be better, I'll look at this, but the issue is that existing users could be quite badly affected by the change as their properties may no longer align, and it would default to the wrong value.

As per offline discussion earlier, I think that we should get the state-machine based encoder working first, and then maybe even create an entirely new plugin item for it. Marking this one as deprecated or similar.

vzahradnik, 2024-05-29:
Yes, I agree. It is not a blocker because users can set this option in the code. This is what I did when I tested the implementation.

davetcc, 2026-10-03:
This is done for the new ISR hardware encoder, but not for the existing one, as it would break sketches.

closed 2026-10-03 by davetcc