TcMenu/tcMenu #471: XPT2046 Touch screen control on other SPI bus

paulvk2, 2024-05-11:
With the ESP32-2432S028 and other similar displays the touch control chip is on the second buss of the ESP32 but tcMenu has no options when the TFT and touch are on different bus.

davetcc, 2024-05-15:
Need more information to understand what plugins you're using etc. 

For example, the TFT-eSPI plugin uses the libraries internal interrogator, and therefore would be configured within that library. So does the BSP touch plugin.

The only other plugin is the ArduinoLibrary based plugin, I am assuming that. However, someone will need to determine how the plugin needs to change in order to pass in other than the default bus, is it in the variable definition, here in the plugin:

https://github.com/davetcc/tcMenu/blob/master/xmlPlugins/core-display/ArduinoTouchLibraryIntegration.xml#L58

paulvk2, 2024-05-17:
If you add GFX_Library_for_Arduino for the ESP devices as selection it will simplify things as many displays are preconfigured for use .
 In the PDQgraphicstest example are Arduino_GFX_databus.h , Arduino_GFX_dev_device.h , Arduino_GFX_display.h , Arduino_GFX_pins.h .  Arduino_GFX_dev_device.h contains a large number of displays by uncommenting the used display the configuration is taken care of and in the TouchCalibration example touch.h just needs the GPIO numbers used
Regards Paul

AtmosphEng, 2024-12-24:
Hi, I've just been trying the same sort of thing e.g. trying the get tcMenu to run on an ESP32 2432S028R Cheap Yellow Display (CYD) using the XPT2046 Touch screen SPI  (VSPI) controller. The GPIO pins are as detailed by paulvk2 above. I've tried all the related examples but I've failed to get any tcMenu input working from the touch screen of the CYD.
Do you know of any working code examples of tcMenu on the CYD with its XPT2046 VSPI resistive touch screen input device? Thanks.

AtmosphEng, 2024-12-26:
A successful workaround for (this) issue https://github.com/TcMenu/tcMenu/issues/471 is in Github repo https://github.com/AtmosphEng/TcmTouch-XPT2046.git. tcMenuDesigner is used to generate the XPT2046 touchscreen as per normal and the workaround code is found by searching for the tag: WORKAROUND_TCMENU_WITH_ESP32_CYD. Basically the ESP32_CYD uses a different SPI for the resistive touchscreen on SPI2 (VSPI). The io pin set for VSPI is put in *_main.cpp and is used to create mySPI. the tcMenuDesigner generated file tcMenuAdaTouchDriver.h is modded to access mySPI via an extern. Iin its init(), theTouchDevice(); is then modded to theTouchDevice(mySpi);.
Thanks to Brian Lough for the initial ESP32_CYD resistive touch code sample at: https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/tree/main/Examples/Basics/2-TouchTest.