TcMenu/tcMenu #472: Problem compiling U8G2_ST7920_128x64

rogi102, 2024-05-18:
Hi
I'm trying to compile code on ARDUINO for an ST7920 display (LCD 12864) after generating the code in TcMenu.
 I get this compilation error:

I:\STM DUINO\TEST_MENU\TEST_MENU_menu.cpp:18:34: error: expected primary-expression before ')' token
   18 | U8g2Drawable tcgfxDrawable(&tcgfx);
      |                                  ^
I:\STM DUINO\TEST_MENU\TEST_MENU_menu.cpp: In function 'void setupMenu()':
I:\STM DUINO\TEST_MENU\TEST_MENU_menu.cpp:30:10: error: expected primary-expression before '.' token
   30 |     tcgfx.begin();
      |          ^

exit status 1

Compilation error: expected primary-expression before ')' token

What's the problem?

davetcc, 2024-05-26:
Please always follow the issue template, either correct this issue to follow the template (which gives enough information for us tio be helpful) or close and create a new one.

davetcc, 2024-06-02:
This issue is lacking enough information to provide any assistance. When raising such issues you need to provide a minimal project that we can look at zipped up using a regular format. You also need to follow the issue template. If you find that you can't follow the issue template, raise as a discussion instead. 

I recommend that you create a simple project that presents the same error, fully document what board and core you are using, the version of the library and designer, and then recreate either a new issue following the template or raise a discussion. Presently this just noise so I am closing it.

closed 2024-06-02 by davetcc