TcMenu/tcMenu #473: SubMenut Callback

tomelgato, 2024-05-20:
**Describe the bug**
In TCMenu Designer 4.2.2 its possible to define a callback for a SubMenuitem but it gets lost after you leave the edit mask for the SubMenuItem

**To Reproduce**
Edit a submenuitem, leave the edit mask, go into edit of the submenuitem again

**Expected behavior**
No callback edit fields for SubMenuItem (imho its not possible to do a callback on SubMenuItem)

**Screenshots - optional**
If applicable, add screenshots to help explain your problem.

**Desktop UI issue (please complete the following information):**
 - OS: mac
 - Version Ventura
 - Designer Version 4.2.2

**Embedded environment - if applicable**
 Provide as much information as possible about the board configuration here.

**Additional context - optional**
Add any other context about the problem here.

davetcc, 2024-05-26:
> No callback edit fields for SubMenuItem (imho its not possible to do a callback on SubMenuItem)

Agreed, it should not be possible, for submenu items the menu observer should be used instead. It should not be possible to edit the field.

davetcc, 2024-05-27:
On reflection, it only needed one line of code adding to make it work in the library and in the UI. Then if someone just wants a simple notification when a submenu is selected it is possible without going the observer approach. As it already almost worked, I've just made it work properly.

davetcc, 2024-05-29:
See library issue https://github.com/davetcc/tcMenuLib/issues/214

davetcc, 2024-05-29:
Will be in the 4.3 release.

closed 2024-05-29 by davetcc