TcMenu/tcMenu #480: Add functionality to save TCMenu settings only after the user completes changing a parameter value

Ales-Svoboda, 2024-07-15:
### Description

Currently, in TCMenu, it is possible to call a callback method when a parameter value is changed, but this method is called every time the value changes. For example, if changing a value from 1 to 100, the callback is invoked 99 times. It would be preferable to save the settings only once after the user has completed the parameter change. Although frequent saving can reduce flash memory lifespan, parameter changes are typically infrequent in many applications. Therefore, an option to enable automatic saving upon completion of a parameter change should be configurable globally for the entire TCMenu.

### Steps to Reproduce

1. Use TCMenu to create a menu with parameters that can be changed.
2. Change a parameter value from a lower number to a significantly higher number.
3. Observe that the callback method is called for each intermediate value during the change.

### Expected Behavior

- The system should call the callback method and save the settings only after the user completes changing the parameter value, not during each intermediate change.
- There should be a global configuration option in TCMenu to enable or disable this behavior.

### Actual Behavior

- The callback method is called for each intermediate value change, which can result in numerous unnecessary save operations.

### Proposed Solution

1. **Modify Callback Logic**:
    - Change/add the logic so that the callback method is called only once after the user has finished changing the parameter value.
2. **Global Configuration Option**:
    - Add a global setting in TCMenu that allows users to enable or disable automatic saving upon the completion of parameter changes.
3. **Implement Save Functionality**:
    - Implement the functionality to save settings automatically after a parameter change is completed.
4. **Test Changes**:
    - Test the new functionality to ensure that it works as expected and that settings are saved correctly without multiple unnecessary saves.

By implementing these changes, TCMenu will provide a more efficient and user-friendly way to handle parameter changes and save settings, improving both usability and the lifespan of flash memory.

davetcc, 2024-07-15:
IMHO, I think that this should be quite straightforward to do using a MenuManagerObserver. You'd only add the extra observer or commit hook if the option was enabled. See:

https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/menumanager-and-iteration/#listening-for-changes-on-menu-manager

Ales-Svoboda, 2024-07-15:
@vzahradnik In our FW we created a button in the menu for manual data saving. Can you remind me why we didn't use the `menuEditEnded() ` method for automatic saving?

vzahradnik, 2024-07-28:
I am sure there was a reason. However I can't tell it with certainty. I will update this ticket once I get back to working with our hardware to check it out.