TcMenu/tcMenu #489: embedControl in designer showing as one black screen

vzahradnik, 2024-08-26:
**Describe the bug**
embedControl inside TcMenu designer is shown as one giant black window

**To Reproduce**
Run the latest stable version of TcMenu designer in Windows. Others systems have to be tested.

**Expected behavior**
I will see the content of the window.

**Screenshots - optional**
[image]

**Desktop UI issue (please complete the following information):**
 - OS: Windows
 - Version Win 11
 - Designer Version and any other relevant details: 4.2.2

davetcc, 2024-08-26:
I think maybe the colors scheme set up is somehow messed up, I will check again how they got migrated from the old format into SQLite, can you try:
[image]

vzahradnik, 2024-08-26:
Your suggestion didn't help. Also I found another bug: Once I change the color scheme and hit Save, after I leave this settings page and go back, the changes are not reflected. I always see just the global color settings without any custom colors.

davetcc, 2024-08-26:
OK, I think the logs may be needed to work out what's going on here.

vzahradnik, 2024-08-26:
I will provide them later.

vzahradnik, 2024-08-27:
I don't see anything unusual in logs but I include them anyway. It is the full output from the moment of starting the designer.

```
[2024-08-27 16:50:30] INFO TccDatabaseUtilities: Checking for table TC_CONNECTION
[2024-08-27 16:50:30] INFO TccDatabaseUtilities: Checking format of TC_CONNECTION
[2024-08-27 16:50:30] INFO TccDatabaseUtilities: Table - TC_CONNECTION, changed= false
[2024-08-27 16:50:30] INFO TccDatabaseUtilities: Checking for table TC_FORM
[2024-08-27 16:50:30] INFO TccDatabaseUtilities: Checking format of TC_FORM
[2024-08-27 16:50:30] INFO TccDatabaseUtilities: Table - TC_FORM, changed= false
[2024-08-27 16:50:30] INFO TccDatabaseUtilities: Checking for table GLOBAL_SETTINGS
[2024-08-27 16:50:30] INFO TccDatabaseUtilities: Checking format of GLOBAL_SETTINGS
[2024-08-27 16:50:30] INFO TccDatabaseUtilities: Table - GLOBAL_SETTINGS, changed= false
[2024-08-27 16:50:30] INFO TccDatabaseUtilities: Checking for table GLOBAL_SETTING_COLORS
[2024-08-27 16:50:30] INFO TccDatabaseUtilities: Checking format of GLOBAL_SETTING_COLORS
[2024-08-27 16:50:30] INFO TccDatabaseUtilities: Table - GLOBAL_SETTING_COLORS, changed= false
[2024-08-27 16:50:30] INFO DatabaseAppDataStore: Get global settings from store
[2024-08-27 16:50:30] INFO TccDatabaseUtilities: Checking for table TC_MENU_SETTINGS
[2024-08-27 16:50:30] INFO TccDatabaseUtilities: Checking format of TC_MENU_SETTINGS
[2024-08-27 16:50:30] INFO TccDatabaseUtilities: Table - TC_MENU_SETTINGS, changed= false
[2024-08-27 16:50:30] INFO JdbcTcMenuConfigurationStore: Loaded all configuration
[2024-08-27 16:50:30] INFO JdbcTcMenuConfigurationStore: Loaded initial designer settings asLoadedConfiguration{changed=false, projectMaxLevel=1, numberBackups=20, registeredKey='', lastLoadedProject='null', lastRunVersion='4.3.0', arduinoOverrideDirectory='null', libraryOverrideDirectory='null', locale='sk_SK', recursiveNaming=false, saveToSrc=true, eepromSized=true, usingArduinoIDE=false, releaseStream=STABLE, currentTheme='lightMode'}
[2024-08-27 16:50:30] INFO JdbcTcMenuConfigurationStore: Getting extra plugins
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Traversing C:\Users\vladi\Klienti\TcMenu\tcMenu\xmlPlugins for plugins
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Plugin xml found in C:\Users\vladi\Klienti\TcMenu\tcMenu\xmlPlugins\core-display
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loading plugins in directory C:\Users\vladi\Klienti\TcMenu\tcMenu\xmlPlugins\core-display
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin dfRobotJoystick.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin dfRobotLcd.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin i2cLiquidCrystalDisplay.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin liquidCrystalDisplay.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin adaArduinoSimpleBuilder.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin adaGfxDisplay.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin u8g2SimpleBuilder.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin u8g2Display.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin tftSpiDisplay.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin ssd1306asciiDisplayQuickStart.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin ssd1306asciiDisplay.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin rotaryEncoderInput.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin upDownSwitchesInput.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin oneButtonSwitchesInput.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin twoButtonSwitchesInput.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin analogJoystickInput.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin analogResistiveTouchScreen.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin ArduinoTouchLibraryIntegration.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin esp32TouchSensor.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin matrixKeyboard.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin EmbeddedJFX.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin adaMbedOled.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin adafruitTrueCpp.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin stm32BspFrameBuffer.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin noInput.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin noDisplay.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin customDrawable.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin customDisplay.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Plugin xml found in C:\Users\vladi\Klienti\TcMenu\tcMenu\xmlPlugins\core-remote
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loading plugins in directory C:\Users\vladi\Klienti\TcMenu\tcMenu\xmlPlugins\core-remote
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin noRemote.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin remoteSerial.xml
[2024-08-27 16:50:30] INFO DefaultXmlPluginLoader: Loaded plugin remoteEthernet.xml
[... cut ...]
[2024-08-27 16:50:42] INFO ScrollUpDownEditorComponent: Updating editor for Rom Choice id(25), status is NORMAL
[2024-08-27 16:50:42] INFO TextFieldEditorComponent: Updating editor for RGB id(26), status is NORMAL
[2024-08-27 16:50:42] INFO MenuSelectButtonEditorComponent: Updating editor for IoT Monitor id(27), status is NORMAL
[2024-08-27 16:50:42] INFO MenuSelectButtonEditorComponent: Updating editor for Authenticator id(28), status is NORMAL
[2024-08-27 16:50:42] INFO TextFieldEditorComponent: Updating editor for Simulator Options id(60001), status is NORMAL
```

davetcc, 2024-08-31:
Please can you use any database tool (Squirrel is a free SQL editor) that supports H2DB and open the database "~/.tcmenu/h2db/tcmenuConfig" then send me the contents of `GLOBAL_SETTING_COLORS`. Otherwise zip up the contents of that directory and email them to me, and I'll open the database to check it myself. I thjink I know what's going wrong, but I need to see what's saved to check.

vzahradnik, 2024-09-01:
I emailed the zip file. However, I don't know where to find the color settings in Designer.

vzahradnik, 2024-09-07:
The issue was reproduced and it occurrs only when users use the light mode. Fix should be easy.

closed 2024-09-07 by davetcc