TcMenu/tcMenu #490: embedControl doesn't support non-ANSI characters

vzahradnik, 2024-08-26:
**Describe the bug**
When I connect into a device with i18n enabled (e.g. non-English language with custom font), the remote connection doesn't properly show the menu characters.

**To Reproduce**
Use an example for i18n TcMenu project and connect remotely into it.

**Expected behavior**
I can properly see the characters as expected. In embedControl there is no need to avoid unicode fonts due to memory constraints, etc.

davetcc, 2024-08-31:
Could you please provide a simple project that has such characters so I can build it and test.
Also, when the project is open in TcMenu Designer, does selecting `Preview` from the menu work? How do the items show there? That will quickly show us where the issue is. It may just be the Java API is using the wrong string decoding, which is simple to fix.

vzahradnik, 2024-09-01:
I will prepare an example for you.

davetcc, 2024-09-07:
I can see the problem here - com.thecoderscorner.menu.remote.protocol.TagValTextParser#readString

https://github.com/TcMenu/tcMenu/blob/f3eac4abaeb86ec940e487c29bf0f78133472ed4/tcMenuJavaApi/src/main/java/com/thecoderscorner/menu/remote/protocol/TagValTextParser.java#L50

It reads a byte at a time, I'll think about how to do this better.

davetcc, 2024-09-07:
Fixed by keeping data as bytes until the parsing is complete.

closed 2024-09-07 by davetcc