TcMenu/tcMenu #491: Remove server plugin

vzahradnik, 2024-08-26:
**Is your feature request related to a problem? Please describe.**
Create a new TcMenu plugin which provides remote server support.

**Describe the solution you'd like**
Allow the plugin to set either IP or DNS domain. Also allow to override this config using build flags, similarly how users can override project name. This is required as users might want to connect to different services based on the build configuration - a staging server for testing and a production server for release.

davetcc, 2024-09-01:
Can you try it @vzahradnik  and see if it works OK for you, I tested it on ESP32 and it built and worked.

vzahradnik, 2024-09-01:
I will try it and report back here. Thanks!

vzahradnik, 2024-09-09:
My first observation is regarding the IP address. Is there a way to use a compile flag instead of hardcoding the ip address in the field?

For example, I want to set one IP address for the debug environment and another one for production environment.

DNS resolver where I could type domain name would be even better but we can explore this feature later.

I will have a detailed look at the plugin during the week.

vzahradnik, 2024-09-11:
Another observation: TcMenu tries to connect to remote even though the IP is not reachable. As part of the optimization, we should implement something like this:

- Check if the network connection is established at all
- If it is, try to connect to the remote server
- If the connection is not successful, implement a timeout of 5 seconds
- After the second unsuccessful connection, double the timeout to 10 seconds, etc... up to 5 minutes (configurable as a global TcMenu variable)
- Once the connection is established, reset the timeout

This solution will - I believe - save some energy when running from battery and will reduce the clutter in logs a lot.
[image]

vzahradnik, 2024-09-11:
Long term, I'd like to have a networking module/library ready to handle this for the whole system.

vzahradnik, 2024-09-11:
TcMenuJavaAPI in Maven Central is still on version 4.2.2 and doesn't have support for this feature.

vzahradnik, 2024-09-11:
I will check the encryption code during the weekend. Besides those small optimization issues, the plugin works great!

davetcc, 2024-09-28:
> My first observation is regarding the IP address. Is there a way to use a compile flag instead of hardcoding the ip address in the field?
>
> For example, I want to set one IP address for the debug [...]

You can now set the address either to be hardwired, variable, or compile flag

davetcc, 2024-09-28:
> TcMenuJavaAPI in Maven Central is still on version 4.2.2 and doesn't have support for this feature.

It is now updated, this was due to the account needing an update.

davetcc, 2024-09-28:
>Alternatively, respect the heartbeat global config and initiate the session not sooner than the HEARTBEAT_INTERVAL value.

For now let's do this, and look into a proper networking library as discussed.

davetcc, 2024-10-19:
I think most of the immediate issues are now resolved by 4.4, I'm going to close for now, and we can open issues for anything new that we find.

closed 2024-10-19 by davetcc