TcMenu/tcMenu #502: Remove example from the repository

vzahradnik, 2024-09-07:
**Is your feature request related to a problem? Please describe.**
The TcMenu example in this repository is unused and at this point outdated. Remove it from the repository.

davetcc, 2024-10-19:
We should also refactor the way that TcMenu Designer handles embedded Java builds. My suggestion is to remove all the complexity as the Java version does not need a code generator, simply it needs a way to export the menu tree that would be immediately usable from the starting point application.

My suggestion is that we:

1. Move the existing example to a new repository named `tcmenu-java-examples` along with all the Java API examples.
2. Better document the java embedded example and make it a starting point rather than the present auto-generated mess.
3. Remove the embedded Java code generator and instead provide a method of getting the MenuTree object directly from designer UI.
4. Provide a "starting point" connectivity UI that can be used by others to build their own Java UI.

davetcc, 2024-10-27:
The work is under way ready for 4.4, we must move it now because there are already critical issues with the dependencies on the embedded java example. The proposed new repo is called `tcmenu-examples-starters` and longer term will look somewhat like

```
java
   apiIsClient
      apiActsAsClientToDevice
      apiActsAcceptsFromDevice
      emebedControlUI
   apiIsServer
      embeddedJavaDeviceUI
      deviceWithClientConnectionExample
   encryption
      makeEncryptionKeyExample
dotnet
   dotNetConnectsToDevice
flutter
   embedControlFlutter
python
   ...
```

closed 2024-11-04 by davetcc